FreeCAD/FreeCAD #13735: Measurement: unsaved measure are a document modification

pierreporte, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

1. Open an existing file and create a new one.
2. Measure something but don’t click on Annotate.
3. The document is marked as modified.
4. Cancel the measurement.
5. The document is still marked as modified.

Indeed, there the measurement in the design tree but it goes away is left unsaved. This can lead to problems when working with assemblies and multiple files because the programs can believe that there are modified files when they aren’t and save them when they shouldn’t. This can be especially annoying when working with a PDM.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Anything else?

On save management of files: #12680.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-30:
Also, a new saved annotation is marked for recompute instantly. Maybe this is related.

pierreporte, 2024-06-28:
@hlorus What’s your opinion on this issue? The behavior is very frustrating.

pierreporte, 2024-10-14:
@maxwxyz since the fix was reverted, do we reopen the issue or a new one should be created to be more generic as @wwmayer explained?