FreeCAD/FreeCAD #13736: opening FC on mac via double click of FC file causes {addons to not be usable, part to render incorrectly}

inieves, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In MacOS, on FC 0.21.2-33771 I have installed addons (e.g. Fasteners, etc.).  Expected behavior is that those addons are usable regardless of how FC is launched.  However, I have discovered that those addons are usable if FC is launched directly (e.g. via app icon or spotlight) but those addons are not usable if FC is launched indirectly (e.g. via double-click on an FC file).

I was able to detect that the addons were not loaded because error messages resulted and the part did not render correctly.  Error messages:

14:06:08  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Users/inieves/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.64a
  * fasteners 0.5.20
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-30:
Is this also the same in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Would a fresh install with no prior mods / config files solve this?
I am unsure if this is related to the addon itself (fasteners) is there an update available in the addon manager?

inieves, 2024-04-30:
Hi Max,

I was using a fresh install, and had never previously installed any mods/configs.  In that sense, I think it was quite a fresh install.

The addons were all latest.  I refreshed my cache and ensured no updates were available.

I slightly think this may be contrained to Fasteners… because I have a dependency on another addon, something called a2p… and that seems to load alright no matter how I restart.

At the moment, I cannot install an unreleased version of FC on my primary workstation, but I can give you a copy of the part/file that caused me failure.  It is attached here, for test purposes only.

￼

maxwxyz, 2024-05-01:
Without mods means without any addons as well, so without this folder `$HOME/Library/Application Support/FreeCAD`
(https://forum.freecad.org/viewtopic.php?style=10&p=618244#p618244)

The test file did not upload here, you can drop in a .zip on GitHub.

inieves, 2024-05-01:
Hi Max,  I attached the file as a .zip here.

[linkage_lock.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15177299/linkage_lock.FCStd.zip)

inieves, 2024-05-01:
I will try to run some of the tests you mentioned later today or later this week as I become more available.

inieves, 2024-05-01:
Hi Max,

“without mods” I receive this error when I double-click my FC file to start FC and also when I first open FC and then open the FC file:

11:20:58  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named ‘FastenersCmd'

I am not experienced with FC enough to know if this is normal/correct behavior.

inieves, 2024-05-01:
Hi Max,

I can verify that on the latest weekly build, the problem exists, exactly as I experienced on the official release.

I used this dev release to test:
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git)
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.64a
  * fasteners 0.5.20

I did not mention, but maybe relevant, I am on ARM not x86

inieves, 2024-05-01:
This is the associated error message:

11:30:24  pyException: Traceback (most recent call last):
  File "/Users/inieves/Desktop/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/inieves/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)

maxwxyz, 2024-05-01:
can you uninstall fasteners addon and try again?

inieves, 2024-05-01:
Hi Max,

I did that uninstall on the dev release.  I receive this error message on both methods of starting FC:

12:42:25  pyException: Traceback (most recent call last):
  File "/Users/inieves/Desktop/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'FastenersCmd'

luzpaz, 2024-08-25:
@inieves please try to reproduce again on latest dev version

luzpaz, 2024-08-25:
@JohnOCFII do you mind testing with fastener wb installed on macos ?

JohnOCFII, 2024-08-25:
> @JohnOCFII do you mind testing with fastener wb installed on macos ?

Behavior is still different (and likely broken) Here's what I see in the console window.  I think there are some unrelated errors with spreadsheet references -- perhaps referring to a different file?  I'd guess those can be ignored.

```
09:54:46  (qt.qpa.fonts) Populating font family aliases took 49 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
09:54:48  pyException: Traceback (most recent call last):
  File "/Applications/Other Apps/FreeCAD-38459.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/osh/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)
09:54:48  pyException: Traceback (most recent call last):
  File "/Applications/Other Apps/FreeCAD-38459.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/osh/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.D6: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vS'
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.C6: unsupported operand type(s) for /: 'str' and 'int'
in expression: D6 / 10
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.D5: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vC'
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.C5: unsupported operand type(s) for /: 'str' and 'int'
in expression: D5 / 10
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.D4: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.C4: unsupported operand type(s) for /: 'str' and 'int'
in expression: D4 / 10
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.D3: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.C3: unsupported operand type(s) for /: 'str' and 'int'
in expression: D3 / 10
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.D2: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vAP'
09:54:48  <Spreadsheet> Cell.cpp(668): linkage_lock_new#Spreadsheet.C2: unsupported operand type(s) for /: 'str' and 'int'
in expression: D2 / 10
09:54:48  <Spreadsheet> Sheet.cpp(1627): Failed to restore linkage_lock_new#Spreadsheet: One or more cells failed contains errors.
```

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.17
  * fasteners 0.5.27
```
[image]

luzpaz, 2024-08-25:
CC @shaise have you seen this behavior before ?

inieves, 2024-08-25:
I am unable to test because the latest MacOS dev version DMG seems to be corrupted (failed verification).
I made 4 attempts.  I am attaching a screenshot of the error message.

If someone can let me know another way to get the latest dev version I can try that.

Best,
Ian

￼

inieves, 2024-08-25:
Here is the screenshot.

[image]

JohnOCFII, 2024-08-25:
> If someone can let me know another way to get the latest dev version I can try that.

No easy way other than building FreeCAD locally (there are instructions if you are interested).  The problem with opening FreeCAD on macOS is actively being worked.

inieves, 2024-08-26:
is there need for access to a mac?  i have a mac mini in cloud that you can access graphically via RDP if necessary.

inieves, 2024-08-27:
Sure kindly let me know where the instructions are.  Also, if the build resulted in corrupted DMG would manually compiling produce a better result?Best,Ian

JohnOCFII, 2024-08-27:
> Sure kindly let me know where the instructions are.  Also, if the build resulted in corrupted DMG would manually compiling produce a better result?Best,IanOn Aug 25, 2024, at 4:29 PM, JohnOCFII ***@***.***> wrote:

Greetings -- Actually, the easiest way to get the latest development release to work on an M1/M2/M3 Mac (until the FreeCAD packaging bug is fixed) is as follows:

1. Download the (actually twice) weekly build from here:  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds - so for this week's Arm Mac it would be: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38553-conda-macOS-arm64-py311.dmg
2. Double-click on the .dmg file.  It will open a small window with the FreeCAD App.  Drag the FreeCAD app to the Application folder by dragging it to the alias of the Application folder (as shown).
3. Open a terminal window.
4. Within the terminal window, navigate to your Application folder:  Should be similar to:
`cd /Applications`
5. Execute the following command: `xattr -cr FreeCAD.app`
6. Run FreeCAD by double-clicking the App icon in the Finder.

The bug will likely be resolved by next weekend if you choose to wait.

inieves, 2024-08-27:
Thank you for those instructions.

I can confirm that the bug still exists in the latest dev version 0.22.0dev on MacOS 14.2.1 on M2

In my latest test I was able to eliminate unrelated failures (caused by associated parts not being present) to produce the relevant error only.  The error message is pasted below:

16:56:11  (qt.qpa.fonts) Populating font family aliases took 143 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Body
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Sketch
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Revolution
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Revolution
16:56:13  pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/inieves/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#ScrewDie
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Cut
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#Sketch001
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#SubtractiveHelix
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#SubtractiveHelix
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#SubtractiveHelix001
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#SubtractiveHelix001
16:56:13  <App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: linkage_lock#PolarPattern
16:56:13  pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/inieves/Library/Application Support/FreeCAD/Mod/fasteners/./FastenersCmd.py", line 26, in <module>
    from FreeCAD import Gui
<class 'ImportError'>: cannot import name 'Gui' from 'FreeCAD' (unknown location)

luzpaz, 2024-08-27:
Is this the mysterious macOS Gui bug ? https://github.com/FreeCAD/FreeCAD/issues/13108#issuecomment-2016424077

luzpaz, 2024-10-01:
@hyarion any chance you can look at this macOS bug ?

hyarion, 2024-10-01:
@luzpaz sure! Can you assign me to it?

hyarion, 2024-10-01:
I've tested this with 1.0.0RC2 and the latest versions of a2plus and fasteners and I don't get the py-exception.
I do only see the missing file errors that was discussed earlier.

@inieves Can you check if this still is an issue? If it still is an issue, can you upload a file where it happens that doesn't contain the missing file issues?

<details><summary>Missing file errors</summary>
<p>
23:04:45  <Exception> File '/Users/nauck/Downloads/linkage_button.FCStd' does not exist!
23:04:45  Exception opening file: /Users/nauck/Downloads/linkage_button.FCStd [File '/Users/nauck/Downloads/linkage_button.FCStd' does not exist!]
23:04:45  <Exception> File '/Users/nauck/Downloads/linkage_hold.FCStd' does not exist!
23:04:45  Exception opening file: /Users/nauck/Downloads/linkage_hold.FCStd [File '/Users/nauck/Downloads/linkage_hold.FCStd' does not exist!]
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D6: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vS'
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C6: unsupported operand type(s) for /: 'str' and 'int'
in expression: D6 / 10
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D5: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vC'
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C5: unsupported operand type(s) for /: 'str' and 'int'
in expression: D5 / 10
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D4: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C4: unsupported operand type(s) for /: 'str' and 'int'
in expression: D4 / 10
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D3: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C3: unsupported operand type(s) for /: 'str' and 'int'
in expression: D3 / 10
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D2: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vAP'
23:04:45  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C2: unsupported operand type(s) for /: 'str' and 'int'
in expression: D2 / 10
23:04:45  <Spreadsheet> Sheet.cpp(1627): Failed to restore linkage_lock#Spreadsheet: One or more cells failed contains errors.
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D6: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vS'
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C6: unsupported operand type(s) for /: 'str' and 'int'
in expression: D6 / 10
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D5: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vC'
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C5: unsupported operand type(s) for /: 'str' and 'int'
in expression: D5 / 10
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D4: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C4: unsupported operand type(s) for /: 'str' and 'int'
in expression: D4 / 10
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D3: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C3: unsupported operand type(s) for /: 'str' and 'int'
in expression: D3 / 10
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D2: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vAP'
23:04:49  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C2: unsupported operand type(s) for /: 'str' and 'int'
in expression: D2 / 10
23:04:49  Input: One or more cells failed contains errors.</p>
</details> 

<details><summary>Full version</summary>
<p>
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * fasteners 0.5.29
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
</p>
</details>

inieves, 2024-10-02:
I have also tested on 1.0.0RC2 and the latest versions of a2plus and fasteners.
I also see only the missing file errors, I believe (see output below)

I believe this error output is related only to missing files.
(I do not have a version of this file with errors but without missing files)

09:56:19  <Exception> File '/Users/inieves/Desktop/linkage_button.FCStd' does not exist!
09:56:19  Exception opening file: /Users/inieves/Desktop/linkage_button.FCStd [File '/Users/inieves/Desktop/linkage_button.FCStd' does not exist!]
09:56:19  <Exception> File '/Users/inieves/Desktop/linkage_hold.FCStd' does not exist!
09:56:19  Exception opening file: /Users/inieves/Desktop/linkage_hold.FCStd [File '/Users/inieves/Desktop/linkage_hold.FCStd' does not exist!]
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D6: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vS'
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C6: unsupported operand type(s) for /: 'str' and 'int'
in expression: D6 / 10
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D5: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vC'
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C5: unsupported operand type(s) for /: 'str' and 'int'
in expression: D5 / 10
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D4: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C4: unsupported operand type(s) for /: 'str' and 'int'
in expression: D4 / 10
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D3: Document '<<linkage_button>>' not found in '<<linkage_button>>#<<Input>>.vP'
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C3: unsupported operand type(s) for /: 'str' and 'int'
in expression: D3 / 10
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.D2: Document '<<linkage_hold>>' not found in '<<linkage_hold>>#<<Input>>.vAP'
09:56:19  <Spreadsheet> Cell.cpp(668): linkage_lock#Spreadsheet.C2: unsupported operand type(s) for /: 'str' and 'int'
in expression: D2 / 10
09:56:19  <Spreadsheet> Sheet.cpp(1627): Failed to restore linkage_lock#Spreadsheet: One or more cells failed contains errors.

hyarion, 2024-10-02:
The initial error cannot be reproduced. Whatever it was causing the issue, it must have been resolved since.

I think we can close this issue now.

maxwxyz, 2024-10-02:
@inieves please comment if this is still an issue

closed 2024-10-02 by maxwxyz

inieves, 2024-10-02:
At this time I cannot recreate the issue.The issue appears to be resolved.