FreeCAD/FreeCAD #13739: Support making measurements of STL and other mesh-based formats

prokoudine, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There seems to be a use case where you might want to make measurements of an actual 3D print against the STL used to make that print. The old measurement tool allow making some rough measurements of STL files, the new one doesn't allow doing that, it seems to only operate on breps.

It looks like it's doable though, e.g. Prusa slicer is capable of detecting various features such as linear and circular edges and thus measure length and diameter:

https://www.youtube.com/watch?v=tYWsjLCo_zU

A quick look suggests that they use functions in libslic3r/Measure.cpp for that. Is it something that FreeCAD could do as well, by either reusing libslic3r (unsure, it's AGPL-3.0) or reimplementing feature detection?

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * ExplodedAssembly
  * sheetmetal 0.4.11
  * BIM 2021.12.0
  * A2plus 0.4.61
  * OpenTheme 2024.4.20
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
Is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/13709 or do you mean adding more abilities as to determine whether the current edge belongs to a loop which forms aholw and the diameter should be approximated?

prokoudine, 2024-05-01:
> or do you mean adding more abilities as to determine whether the current edge belongs to a loop which forms aholw and the diameter should be approximated?

That is precisely it.