FreeCAD/FreeCAD #13744: [Regression] Rework initial dialog when creating a Sketch in PartDesign

chrisb-github, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting something from outside of a body and then creating a sketch in PartDesign, the user is shown a dialog with three options.

This should be reworked for several reasons:
- The recommended option is not working (regression)
- The recommended option is not parametric
- The third option creates a faulty model, it should be dropped.

There should be only two options, and the other should be the default, i.e. the options should be:

- Make dependent additional shape binder (recommended)
- Make independent additional shape binder

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git)
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.15
  * ExplodedAssembly
  * Curves 0.6.29
```

### Subproject(s) affected?

None

### Anything else?

See https://forum.freecad.org/viewtopic.php?t=87230

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-05-01:
I would add that if possible at all, it would be helpful to have a tooltip for each option that explains it in more detail.

maxwxyz, 2024-05-01:
SubShapeBinder should be the tool used, otherwise I agree.

jmickelin, 2024-05-23:
A good suggestion, but to me the phrasing sounds a bit clunky and would cause a bit of head scratching.

When I read "Make dependent additional shape binder", the word order "dependent additional" suggests that the "additional" is somehow a property of the "shape binder" to be created. As in, "what the heck is an additional-shape-binder?".

This might be case of the [implicit English adjective word order](https://www.theguardian.com/commentisfree/2016/sep/13/sentence-order-adjectives-rule-elements-of-eloquence-dictionary) making the phrasing sound cumbersome.

I think it flows better as "Make (an) additional dependent/independent shape binder". Or even better -- since "additional" also carries some connotations of creating more than one thing (though I suppose that "other thing" would be the sketch itself) -- just use a more neutral word. Something like "Make a new dependent/independent shape binder" or even "Create (a) new dependent/independent shape binder".