FreeCAD/FreeCAD #13745: Tool path leak

KR-Huck, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Tool path briefly cuts outside of pocket with adaptive profiling. 

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]
[arleft.zip](https://github.com/FreeCAD/FreeCAD/files/15172671/arleft.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
Could you try this with the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

KR-Huck, 2024-05-01:
Thanks for such a fast response. This fixed the problem, no more tool path leaks! FreeCad is a great program.

closed 2024-05-01 by maxwxyz