FreeCAD/FreeCAD #13746: Variant links to lattice2 arrays fail with Property container has no property 'Support'

johnsonm, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a project which no longer functions correctly: https://gitlab.com/mcdanlj/RSVPGameReplacementParts

When I open CubeSet.FCStd from that project, I get repeated errors:
```
21:38:39  pyException: Traceback (most recent call last):
  File "/var/home/johnsonm/.local/share/FreeCAD/Mod/lattice2/./lattice2BaseFeature.py", line 278, in onDocumentRestored
    self.disableAttacher(selfobj)
  File "/var/home/johnsonm/.local/share/FreeCAD/Mod/lattice2/./lattice2BaseFeature.py", line 272, in disableAttacher
    selfobj.setEditorMode(prop, 0 if enable else 2)
<class 'AttributeError'>: Property container has no property 'Support'
```

I have updated to the latest Lattice2 and the problem persists.

Commenting out the `'Support'` from `attachprops` resolves the error message, but also the variant links are not working correctly; I don't know whether that's because I created the links wrong, or it's a bug here.

I don't know whether the problem is that Lattice2 needs to be adjusted to current FreeCAD or that FreeCAD broke compatibility with Lattice2 unintentionally. If I am reporting this in the wrong place, please let me know.

### Full version info

```shell
OS: Fedora Linux 40.20240419.n.0 (Silverblue) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git) AppImage
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-05-01:
I have at least a workaround for propagating changes here. Changing the links to Tracking, then touching the configuration (without changing it), then explicitly recalculating, for all of the links, allowed me to use my model. Just being clear here that the bug report is **only** meant to be for anything identified as **actually** a bug and not for PEBCAK. ☺

The https://wiki.freecad.org/Variant_Link documentation does not appear to be explicit about how to accomplish my task, so I'm working a bit in the dark here.

maxwxyz, 2024-05-01:
Did you create the issue on the lattice2 repo as well?
Could you test if it works in the stable 0.21.2 version of FreeCAD?

johnsonm, 2024-05-02:
I originally created it in FreeCAD_0.21.2-2023-12-17-conda-Linux-x86_64-py310.AppImage or FreeCAD_weekly-builds-35966-2024-02-14-conda-Linux-x86_64-py310.AppImage (using file and commit dates on my system) and wanted to know if this was an intentional change that should force a lattice2 change or an accidental impact on lattice2 before creating a ticket there.

It wasn't clear to me whether something created with these tools in the development version is even expected to work on the older stable release. Are you confident that it should?

krushia, 2024-05-02:
https://github.com/FreeCAD/FreeCAD/issues/12894 aka https://github.com/FreeCAD/FreeCAD/pull/12714#issuecomment-1983318226
(affects me too)

johnsonm, 2024-05-02:
Thanks for the link that makes it clear that this is an intentional change to fix a but. Closing this report in favor of the Lattice2 issue I just filed at https://github.com/DeepSOIC/Lattice2/issues/86 since this will need a workaround there, and there is no change to make in FreeCAD proper.

closed 2024-05-02 by johnsonm

johnsonm, 2024-05-05:
PR at https://github.com/DeepSOIC/Lattice2/pull/87 for anyone else who finds this issue in a search. ☺