FreeCAD/FreeCAD #13749: Add the ability to clear recent documents list

furgo16, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes the list of recent files can become cluttered, particularly if the number of files shown is set to be reasonably high. This is a feature request to provide the ability to clear that list. Potential use cases:

- The user has finished working with a multiple files project, moves on to the next project and wants a clean slate.

An example implementation from a CAD package (QCAD):
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git) Snap 1002
Build type: Release
Branch: main
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
please add the comments to https://github.com/FreeCAD/FreeCAD/issues/12842 this should be fixed together.

maxwxyz, 2024-05-01:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12842

closed 2024-05-01 by maxwxyz

furgo16, 2024-05-01:
Thanks, I don't mind either way. Alternatively, the two issues on #12842 could be written as a [GitHub task list](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/about-task-lists#about-issue-task-lists), with the first one pointing to this one.