FreeCAD/FreeCAD #13750: Arch: error when setting the Terrain property: gp_GTrsf::Trsf() - non-orthogonal GTrsf

furgo16, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open attached file
2. In the tree view, choose the Site component
3. Set the Site's Terrain property to the Terrain solid
4. The following error is shown, and the Site component displays a warning icon:

[image]

```python
07:39:24  pyException: Traceback (most recent call last):
  File "/snap/freecad/1002/usr/Mod/Arch/ArchSite.py", line 678, in execute
    shape = shape.transformGeometry((shape.Placement * pl).Matrix)
<class 'Part.OCCError'>: gp_GTrsf::Trsf() - non-orthogonal GTrsf
07:39:24  Site: gp_GTrsf::Trsf() - non-orthogonal GTrsf
```

Forum discussion: https://forum.freecad.org/viewtopic.php?p=756278#p756278. I've also been pointed out to https://github.com/FreeCAD/FreeCAD/issues/9651 as a possible cause, but I cannot tell whether it applies.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git) Snap 1002
Build type: Release
Branch: main
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Arch

### Anything else?

1. File to reproduce: 
[Site-loft.zip](https://github.com/FreeCAD/FreeCAD/files/15174094/Site-loft.zip)
1. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-05-02:
When I have encountered error before, it was because the shape as not "triangulated".   Making a copy of the shape seems to restore/create the triangulation.

            // copying the shape prevents "non-orthogonal GTrsf" errors in some versions
            // of OCC.  Something to do with triangulation of shape??
            // it may be that incremental mesh would work here too.
            BRepBuilderAPI_Copy copier(shape);
            ts = Part::TopoShape(copier.Shape());

Roy-043, 2024-05-02:
The code already creates a copy of the shape. This is truly a duplicate. Closing it as such.

closed 2024-05-02 by Roy-043

furgo16, 2024-05-02:
Thanks. I assume a duplicate of https://github.com/FreeCAD/FreeCAD/issues/9651.

furgo16, 2024-07-14:
I can no longer reproduce this issue, even though the OCCT version has stayed the same (7.7.1)

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38126 (Git) Snap 1074
Build type: Release
Branch: main
Hash: ac2b65c37b18617101822e5ece06f3c9b21375cd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

likettly, 2024-10-21:
i am on OCC 7.7.x ，by using BRepBuilderAPI_GTransform encountering this problem,