FreeCAD/FreeCAD #13751: DXF import needs a dedicated import dialog

furgo16, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, when either opening (File > Open) or importing (File > Import) a DXF file, and if the "Show this dialog when importing and exporting" setting under Edit > Preferences > Import-Export > DXF has been checked, that same preferences dialog is shown:

[image]

While effective for code reusability, the user experience is confusing::

1. why should the preferences dialog pop up if I only want to import?
2. if I change the import settings for the file I'm importing, will it change those settings globally?
3. I'm shown more options than I need (I can choose any setting from unrelated sections)
4. this can lead to 

This issue proposes to create a dedicated DXF import dialog that is separate from the DXF Import-Export settings page:

1. It exposes the same options, except for "Show this dialog when importing and exporting". That setting should be in the preferences page only.
2. It should only pop up when invoking File > Open or File > Import and the chosen file in the file picker is a DXF document
3. It should have an OK and Cancel button. OK: go ahead with import, Cancel: cancel operation and do not import anything.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git) Snap 1002
Build type: Release
Branch: main
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

File formats

### Anything else?

For completeness and consistency, other importer's preference should follow the same concept of dual import/settings dialogs. However, the only other importers that have the "Show this dialog when importing and exporting" setting are IFC import and IFC export. If I understand it correctly, they will go away in the near future, so it might not make sense to apply changes to them.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
related: https://github.com/FreeCAD/FreeCAD/issues/13599

kpemartin, 2024-09-18:
There may be some options from other tabs of the preferences dialog that affect the results of the import. Actually, not so much for Import, but for Open, where some options could affect drawing defaults.
Unfortunately this is a somewhat scattered assortment and difficult to characterize. Options that can be changed after the new drawing is created but do not affect the import itself would not be needed. Whether others have an effect depends on other import options, for instance, the standard text size in Draft-> Texts and dimensions is only relevant if the "Use standard font size for texts" option in Import-Export -> DXF is selected. Another example, affecting both Open and Import, is the colour remapping that occurs to prevent objects from being too close to the background colour; this depends on Display -> Colors [Background color] which is an environmental attribute (not a drawing attribute) but permanently affects imported colours.

yorikvanhavre, 2024-09-19:
I agree generally. That's what I did now with the NativeIFC importer. The general preferences page is meant to have all the options, an is not very suited to pop up at each import. We should have another, smaller, quicker dialog that can be popped up more easily. Although it might look like duplicated work, it's also actually a good opportunity to rethink things more frequently and keep the options set small.

But basically, more than that, the whole DXF preferences system needs to be rewritten.

Basically I'd suggest assuming the C++ importer is the standard now, and redesign an options dialog based on it. Then we could check what options are still used by the legacy one and see what we do with them

closed 2025-07-28 by yorikvanhavre