FreeCAD/FreeCAD #13752: Bring back additional folder on the Start page

furgo16, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After reworking the Start page in recent builds, the ability to specify a custom folder and show its contents on the Start page has not been reinstated. This issue advocates for bringing back the ability to define a custom folder and to display its contents on Start.

[image]
_Original option to define the additional folder on FreeCAD 0.21's Start page_

Using Linux packages, this is useful for a few reasons:

1. The Examples folder location is hardcoded as far as I can tell. Probably the same on all platforms?
1. Being able to set the Examples location, either by making its path editable in Preferences, or by extending it (making a Custom location editable) provides more flexibility. In some cases, it's not only about flexibility, but being able to control the contents displayed on the Examples section of the Start page at all. See points below.
1. The .deb package installs examples on `/usr/share/freecad-daily/examples/`, which is only editable by root. Therefore non-root users cannot modify the Examples folder.
1. The snap package installs examples on `/snap/freecad/current/usr/share/examples/`, which is read-only. Not even root users can modify the Examples folder there.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git) Snap 1002
Build type: Release
Branch: main
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

- Forum discussion: https://forum.freecad.org/viewtopic.php?p=756445#p756445

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
Is the issue really about another folder location or just use a proper location for examples on every OS? I guess this needs more discussion.

furgo16, 2024-05-01:
Good point. The issue's intention is to advocate for the ability to define an additional folder. Perhaps the limitations of not having one on Linux could be addressed on a separate issue, but mainly this one is for having an independent additional folder.

closed 2025-02-17 by chennes