FreeCAD/FreeCAD #13754: FemCalculixCantilever3D example opens with errors

hybridherbst, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening the "FemCalculixCantilever3D" example results in console errors:
```py
12:43:01  (qt.qpa.fonts) Populating font family aliases took 139 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
12:43:19  Cannot create object 'Box_Mesh': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'FemConstraintFixed': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'FemConstraintForce': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'Analysis': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'CalculiXccxTools': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'WarpVector': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'Pipeline': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'ResultMesh': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Cannot create object 'CCX_Results': (dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/Fem.so, 0x0002): symbol not found in flat namespace '___kmpc_critical')
12:43:19  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femguiutils/migrate_gui.py", line 163, in exec_module
    return self.load_module(module)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femguiutils/migrate_gui.py", line 210, in load_module
    import femviewprovider.view_material_common
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femviewprovider/view_material_common.py", line 37, in <module>
    from . import view_base_femconstraint
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femviewprovider/view_base_femconstraint.py", line 35, in <module>
    from femviewprovider import view_base_femobject
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Fem/femviewprovider/view_base_femobject.py", line 37, in <module>
    import FemGui  # needed to display the icons in TreeView
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/FemGui.so, 0x0002): symbol not found in flat namespace '___kmpc_critical'
```

### Full version info

```shell
[code]
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Affected: example project

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
Please try this in the current weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-07-13:
@JohnOCFII, @ipatch can y'all reproduce this on dev version ?

JohnOCFII, 2024-07-13:
> @JohnOCFII, @ipatch can y'all reproduce this on dev version ?

Opens fine for me:

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.3
```

[image]

luzpaz, 2024-07-13:
Thanks! Closing as it's fixed in dev version.

closed 2024-07-13 by luzpaz