FreeCAD/FreeCAD #13756: Embedded Text Document modified state isn't correctly propagated to FC file

0penBrain, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce :
 * Create new document
 * Add an embedded Text Document
 * Save the file to reset modified state
 * Enter some text in the Text Document editor => :warning: The Text Document tab indicates a modified state (`*` is added) but it's not propagated to FC file -- 3D view tab --).

Try same with a SpreadSheet instead of a Document Text to see the difference.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37076 (Git)
Build type: Debug
Branch: master
Hash: 02df1c86b70b155886e64a4ad7f944ed9cd3582d
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-18:
Loosely related: #13757

mwganson, 2024-10-20:
Possibly fixed with https://github.com/FreeCAD/FreeCAD/pull/17240

closed 2025-07-09 by github-actions[bot]

0penBrain, 2025-07-11:
Confirmed fixed