FreeCAD/FreeCAD #13757: Embedded Text Document isn't correctly saved when saving FC file

0penBrain, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Create a new document
 * Add an embedded Text Document
 * Save the file to reset the modified state
 * Enter some text in the Document Text editor
 * Go to the file 3D view tab
 * Save the file => :warning: The modified state indicator `*` has been reset on the Text Document editor tab (as if it was saved), but if you try to close the file, you'll get an "unsaved file" notification. If you choose "discard" at this moment, the text in the Text Document won't actually be saved.

You can try using a SpreadSheet instead of the Text Document to see difference.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37076 (Git)
Build type: Debug
Branch: master
Hash: 02df1c86b70b155886e64a4ad7f944ed9cd3582d
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-15:
This seems important as users may rely (critical descriptions an other text) on this feature and then it fail for them.

mwganson, 2024-10-20:
Possibly fixed with https://github.com/FreeCAD/FreeCAD/pull/17240

0penBrain, 2024-11-08:
> @0penBrain could you please test if this is fixed in the latest weekly?

I don't know in weekly but tested with freshly compiled master and can confirm bad behavior is gone.
```
OS: Ubuntu 22.04.4 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Debug
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

closed 2024-11-08 by maxwxyz