FreeCAD/FreeCAD #13758: Document Move Recovery: `XLink` with `UUID`

Andrei-Pozolotin, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a request to resolve the "linked document move" issue.

Currently, a linked document is referenced by a relative file path, 
for example, a `worker.FCStd` a could have a reference to `master.FCStd` as follows:
```xml
<XLink file="master.FCStd" stamp="2024-05-01T10:43:54Z" name="..."/>
```
When`master.FCStd` is moved to a different name/location, and `worker.FCStd` is opened,
then `FreeCAD` merely fails with an error message, while offering no remedy to the situation:
```python
<Exception> File '/.../master.FCStd' does not exist!
```

Since every `FreeCAD` document already has a  `UUID`:
```xml
<Document SchemaVersion="4" ProgramVersion="0.22R37076 (Git)" FileVersion="1" StringHasher="1">
    <Properties Count="16" TransientCount="3">
        <Property name="Uid" type="App::PropertyUUID" status="16777217">
            <Uuid value="3d21b268-4bc3-44b3-8725-1c4c951e8933"/>
```

A better approach to deal with file relocation error situation could be:
* extend `XLink` attributes to include a file `UUID` property
* on `worker` load, on "missing `master` file" condition
* attempt to resolve the `master` file `URL` by `UUID`
* via some resolution mechanism `api`

There could be many different resolution providers: local, remote, python macro, etc,
with the default provided by `FreeCAD` suggested to be  a "project repository assumption".

`Project Repository Assumption` means:
* assume that project `*.FCStd` files are part of  a `git` repository
* external `*.FCStd` file trees could also be grafted here as `git` modules
* then use `git` repository root: `git rev-parse --show-toplevel` as the search root
* and resolve file location by scanning all `*.FCStd` in repository/modules by looking for the `UUID`

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37076 (Git)
Build type: Debug
Branch: main
Hash: 02df1c86b70b155886e64a4ad7f944ed9cd3582d
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-05-01:
and FreeCAD should finally allow the user to find the file by itself i.e. open a file dialog.

maxwxyz, 2024-05-01:
duplicate https://github.com/FreeCAD/FreeCAD/issues/9899 ?

Andrei-Pozolotin, 2024-05-01:
@maxwxyz :  

no, this is not a duplicate of:
* #9899

this is much more basic issue, for example:
* [How to move and rename documents while maintaining links?](https://forum.freecad.org/viewtopic.php?style=4&t=79676)

Andrei-Pozolotin, 2024-05-02:
@sliptonic Brad: 

Perhaps you see a better way to handle "document relocation" issue?

And, assuming that proposed functionality: `Xlink` + ` UUID` + resolution `api` is already there,
would it be of interest for Ondsel to contribute to `FreeCAD` a remote file resolution `provider` for this?

sliptonic, 2024-05-02:
@JohnAD  might want to weigh in or bookmark this issue.  I see it being related to discussions on the topic of USD

sslupsky, 2024-11-30:
Does realThunder's fork provide a way to update links?