FreeCAD/FreeCAD #13760: Qt6: Rendering of overlay TreeView is broken

kadet1090, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When building against Qt6 (which now is forced for Arch linux users) tree view background is rendered for every column of the tree even for indentations:
[image]

This is not supposed to happen. This regression is caused by this commit in Qt: https://github.com/qt/qtbase/commit/16e0e7c779fea47c52a291f175654e6929fa8987. Long story short, until Qt 6.5 Qt did not render items for non-alternating backgrounds, leaving that task for delegates. Delegate for rendering tree items relied on this behavior and now it is broken.

I don't see any easy way around that issue as it is not guarded by any `if` statement that could be abused. My only idea is to somehow override behavior of the 
```
            style()->drawPrimitive(QStyle::PE_PanelItemViewRow, &opt, painter, this);
```
to be no-op but I don't know how to achieve that. One way would be to use `QProxyStyle` to override this but this is not compatible with QSS which renders this solution useless.

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37065 +1 (Git)
Build type: Debug
Branch: more-at-the-end
Hash: 0590b51dabb5380ff983b4df2b7cb2cb21db2d18
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * SelectorToolbar
  * ExtremeProDark 2.9.0
  * BIM 2021.12.0
  * CurvedShapes 1.0.4
  * IconThemes
  * sheetmetal 0.3.0
  * OpenTheme 2024.4.20
  * Glass (Disabled)
  * lattice2 1.0.0
  * A2plus 0.4.60n
  * Curves 0.6.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-05-01:
I don't get this background. Why not remove it?

kadet1090, 2024-05-01:
> I don't get this background. Why not remove it?

Readability, models can have different color, have multiple lines etc so only way to ensure that tree items are easily readable in all scenarios is to provide background for items that will have good contrast against text and hide distracting stuff like lines fighting for attention around text.

MisterMakerNL, 2024-05-01:
Thats why Realthunder added the text effect. 
If applied gently it's hardly vissible, but it does work.

kadet1090, 2024-05-01:
It does not work well enough as with busy model it still is not readable enough, at least for me.

MisterMakerNL, 2024-05-01:
Could you try the new added overlay stylesheets?
I had no issues, but also don't think much people tried it.

kadet1090, 2024-05-01:
People have different tolerance for things being readable or not, depending on various factors. Realthunder implemented it (but remember, that he also implemented other options - including one with background) in the same way as movie captions are done - white text with black outline so it should be readable in every scenario. Unfortunately this effect is mostly effective with big text (bigger outline separating text from the rest of the picture), while we deal with small fonts. In our case we can also get a lot of lines with similar thickness in small area making it very busy and so needing more brain power to process it, requiring to focus on that particular area instead of being able to quickly scan it. I don't mind making the background optional - if for somebody background is distracting (which totally could be the case), I do however mind taking away option of having it.

This is not a thread for this discussion anyway, so let's not go off-topic anymore.

kadet1090, 2024-05-02:
I managed to fix the issue, writeup on how: https://stackoverflow.com/questions/78414383/qt6-disable-drawing-of-default-background-for-qtreeview-items/78421604#78421604

commit b840e1c9 (2024-06-03): Gui: Fix document tree background rendering with overlay (Qt6)

This aims to fix rendering of tree view items in Qt6. While I don't
belive that this is a good way to fix this, I am worried that it is the
only way to do ir.

BC BREAK: This change introduces artificial QTreeView widget that can be
targeted using QSS and can be used in the delegate for painting background of
items. `QTreeView::item` would now be used to render background for the
whole row, while each cell can be targeted using `#DocumentTreeItems`
selector.

More details on implementation:
https://stackoverflow.com/questions/7841 [... cut]
closed 2024-06-03 by chennes