FreeCAD/FreeCAD #13762: PartDesign:Body can't be used as a body BaseFeature

NomAnor, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part features can be used to create a Body with a BaseFeature inside by selecting them and then creating a Body. This allows to change between Part and PartDesign workflow.
It is not possible to create a Body with another Body as a BaseFeature, it is explicitly forbidden in the code:
https://github.com/FreeCAD/FreeCAD/blob/02df1c86b70b155886e64a4ad7f944ed9cd3582d/src/Mod/PartDesign/Gui/CommandBody.cpp#L126-L129

It would be nice to allow this so I can create a base body and then create multiple bodies each with a BaseFeatures pointing to the base body and add different additional featues to them.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 +8 (Git)
Build type: Debug
Branch: pd-trans-body
Hash: 19a1b3fab7c9297c28c3accf2a5fdbf060a83185
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.60.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.0.0
  * Assembly3 0.11.3
  * QuickMeasure 2022.10.28
  * fasteners 0.4.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

Manually changing the 'Base Feature' property of the BaseFeature to point to a Body seems to work without a problem. I don't see a reason why this would not be allowed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-01:
As workaround you could use the shapebinder.

Roy-043, 2024-05-03:
This link describes how it can be done:
https://wiki.freecad.org/PartDesign_Body#Base_Feature

NomAnor, 2024-05-03:
I doesn't, it explains how to use a Part::Feature as a base feature. PartDesign::Body can't be used as a base feature.
> Note: you can't select an existing Body, or any of its [features](https://wiki.freecad.org/PartDesign_Feature), when pressing Body.

A ```ShapeBinder``` can be used as a base feature but I think for a user this is an unnecessary step.

0penBrain, 2024-05-03:
Actually it's possible to do it. Let's say you want Body_A to be the base feature of Body_B (and supposing Body_A already exists).
 * Create a new body Body_B
 * Select Body_B
 * In the Data tab of Property Editor, select "Base Feature" and click on the '...' icon
 * Select "Body_A" in the list and validate
Done :smile: