FreeCAD/FreeCAD #13767: CI Windows Conda C++ Test Failure

oursland, 2024-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Windows Conda C++ Tests are failing. Windows C++ tests were disabled because they failed, but they did succeed under conda until #13333 was merged.  This PR introduced the first dependency upon python's import of FreeCAD modules for the C++ tests.   This line is the offender: 

[`Base::Interpreter().runString("import Draft");`](https://github.com/FreeCAD/FreeCAD/pull/13333/files#diff-61be62ad25b8c9fe4974e6248a58d56ee37fd086062264ef157f0bf9e8733881R370)

For whatever reason, this import is not found and a C++ exception is thrown.  Possibly an issue with `PYTHONPATH` or an incorrect interpreter being selected on Windows.

Should C++ tests be disabled for Windows Conda as it is for Windows to prevent failures on every PR until the correct python path/interpreter can be configured for Windows Conda C++ tests?

@adrianinsaval 

### Full version info

```shell
CI on `main`
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-05-01:
The problem is with the design of those tests: I've disabled them in the interim, while we work on a redesign.

oursland, 2024-05-02:
I'll close the issue, as CI is now functioning again.

closed 2024-05-02 by oursland