FreeCAD/FreeCAD #13774: Aggregate functions should work on lists

krushia, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Aggregate functions](https://wiki.freecad.org/Expressions#Statistical_/_aggregate_functions) appear to only work on spreadsheet ranges an singularly typed items. The problem is easily shown by putting `average(list(1; 2; 3))` in an expression. This cripples the ability to migrate array/lattice parametric wizardry from a spreadsheet to a Property list.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * DynamicData 2.62.0
  * Assembly4 0.50.12
  * Help 1.0.3
  * slic3r-tools
  * ToolbarStyle (Disabled)
  * toSketch 1.0.1
  * sheetmetal 0.4.11
  * ShortCuts (Disabled)
  * fasteners 0.5.17
  * SearchBar 1.0.2
  * Nodes 0.1.36
  * workfeature
  * offline-documentation 1.0.0-alpha
  * Freecad-Built-in-themes-beta 1.2.2
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * freecad.gears 1.2.0
  * Launcher
  * CommandPanel
  * CubeMenu
[/code]
```

### Subproject(s) affected?

Expressions

### Anything else?

[aggregate-functions-not-working-on-lists.zip](https://github.com/FreeCAD/FreeCAD/files/15182465/aggregate-functions-not-working-on-lists.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-02:
@pieterhijma maybe also relevant for VarSets.

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/7b7fab57ee93b5395a24b972503922e255dcb9f8

Aggregate functions also accept lists and tuples, so average(list(1; 2; 3)) works.