FreeCAD/FreeCAD #13788: Using the Help module to show What's this is broken

wwmayer, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Use the command `Std_WhatThis` and click on any active command. No Help window is opened but an error is raised that WebGui isn't an existing module.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git)
Build type: debug
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Anything else?

By default the Help module should use the system browser to show the documentation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-02:
on windows it opens the default browser

maxwxyz, 2024-05-02:
so the solution would be to remove the setting and use the OS browser?

yorikvanhavre, 2024-05-02:
The webGui is still used as a fallback: https://github.com/FreeCAD/FreeCAD/blob/e4eb8598495fa6e3d470ea700eb928c42aaefafe/src/Mod/Help/Help.py#L242

I'll fix that to use the desktop browser instead.

PR #13812 (2024-05-06): Help: Removes use of WebGui
If PySide webenginewidgets is not available, the Help module now defaults to opening the doc in the desktop browser.

Fixes #13788
commit 0262d56e (2024-05-06): Help: Removes use of WebGui (#13812)

* Help: Removes use of WebGui - fixes #13788

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2024-05-06 by sliptonic