FreeCAD/FreeCAD #13789: Measurement: Elements not rendered correctly if source object changes

wwmayer, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Part box
2. Measure the distance between two adjacent vertexes
3. Click on Annotate and close the measurement dialog
4. Select the box and change its length, width, height
5. The label of the distance shows the new value but the displayed positions of the Element1 or Element2 property are wrong

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git)
Build type: debug
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-02:
is this related to https://github.com/FreeCAD/FreeCAD/issues/13718 or even a duplicate? I mean the other issue is about moving but the error is the same, maybe because there is no proper attachment for the position of the annotations (https://github.com/FreeCAD/FreeCAD/issues/13717)

wwmayer, 2024-05-02:
> Is this related to 13718 or even a duplicate?

Yes, it sounds like a duplicate.

maxwxyz, 2024-05-02:
I'll close this then and add a comment with your case

closed 2024-05-02 by maxwxyz