FreeCAD/FreeCAD #13797: Draft: snap intersection hangs indefinitely freecad

alafr, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Open attached file
View > Fit All
Enable Draft snap on intersection
Draw a line and approach the shape
Freecad hangs indefinitely

[snap intersection bug.zip](https://github.com/FreeCAD/FreeCAD/files/15191406/snap.intersection.bug.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37094 (Git)
Build type: Release
Branch: alaf-test
Hash: 779c8949d37ae996cc59a11d373fa518b8a0d094
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * ArchTextures
  * BIM
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * DynamicData 2.60.0
  * Trails 2022.1.0

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * ArchTextures
  * BIM
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * DynamicData 2.60.0
  * Trails 2022.1.0
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

The problem happens in `edge1.distToShape(edge2)` [here](https://github.com/FreeCAD/FreeCAD/blob/e4eb8598495fa6e3d470ea700eb928c42aaefafe/src/Mod/Draft/draftgeoutils/intersections.py#L118)
Snap works fine if I replace
```
        dist, pts, geom = edge1.distToShape(edge2)
        sol = []
        if round(dist, precision()) == 0:
            for p in pts:
                if p[0] not in sol:
                    sol.append(p[0])
        return sol
```
with
```
        sol = []
        section = edge1.section(edge2)
        for vertex in section.Vertexes:
            sol.append(vertex.Point)
        return sol
```
Of course fixing `distToShape `would be better

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-05-02:
This is not a Draft issue as such. I suspect it is the degree of the B-splines (the edges are B-splines with a degree of 1) that is the culprit.

alafr, 2024-05-02:
Those lines are a section between a plane and a ruled surface. Here is there parametric history:
[snap intersection bug history.zip](https://github.com/FreeCAD/FreeCAD/files/15194708/snap.intersection.bug.history.zip)

Roy-043, 2024-05-03:
You are right. The number of knots is large, that may be the cause instead.
`sub.Curve.NbKnots => 947`

PR #15612 (2024-07-29): Draft: Fix snap intersection hang
Fixes #13797.

The code as suggested by @alafr works fine and has basically been implemented.

Code has also been checked against #15350.
commit a1a6f180 (2024-07-29): Draft: Fix snap intersection hang (#15612)

Fixes #13797.

The code as suggested by @alafr works fine and has basically been implemented.
closed 2024-07-29 by yorikvanhavre