FreeCAD/FreeCAD #13801: 3Dconnection Space Pilot stopped working

ShamanTcler, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My Space Pilot worked and then did not work. Now granted this is an older device. The failure mode is, it is just not recognized by Freecad, I can run the working version, stop, load the non-working, and repeat. As the machine is not rebooted seems isolated to the application.

My guess is it is from this change noted in the blog.

WIP Wednesday 17 April 2024: "pskowronskiTDx contributed code to better support 3Dconnexion devices such as SpaceMouse on Windows using Navlib."

Works:
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.35
* lattice2 1.0.0

Fails:

OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37083 (Git)
Build type: Release
Branch: main
Hash: 502f78e111c761cb92c51222605b26b8701fcbca
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.35
  * lattice2 1.0.0

### Full version info

```shell
Works:
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36731 (Git)
Build type: Release
Branch: main
Hash: ac1d887336826c59321352f49a195bf1ef8be2f0
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.35
* lattice2 1.0.0

Fails:

OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37083 (Git)
Build type: Release
Branch: main
Hash: 502f78e111c761cb92c51222605b26b8701fcbca
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.35
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-03:
@pskowronskiTDx could you have a look of this is related to your PR?

pskowronskiTDx, 2024-05-06:
@maxwxyz Please take a look on the initial thread where similar issue has been already reported:  https://github.com/FreeCAD/FreeCAD/pull/12929#issuecomment-2068927980

ShamanTcler, 2024-05-06:
I have known these were old, I have had it since it was new. But it still works.

I gave ordered a new one should be here in a week. Cost was about $150  US

pskowronskiTDx, 2024-05-06:
@ShamanTcler Have you tried with installing the latest 3dxWare (min. v10.5.6) ?

ShamanTcler, 2024-05-06:
I played around with their drivers for a couple of weeks about 2 years ago. Finally got something to work on ONE machine I have two Win 11 machines. 

[image]

This is what ended up working. Anything else did not.

I will try the newer drivers and try to swap back and forth.... 

Nobody really competes with 3DConnexion

maxwxyz, 2024-05-06:
I'll close this then as related to outdated hardware/drivers.

closed 2024-05-06 by maxwxyz

andreymrt, 2024-05-14:
@pskowronskiTDx I'm running into this problem too, but with recent hardware.
I was previously on 3dx 10.8.12 and I updated to 10.8.20 after seeing this post, that didn't resolve the problem

[image]

pskowronskiTDx, 2024-05-15:
@andreymrt This is expected. SpaceMouse settings are no longer available in FreeCAD preferences. You can reach these settings in 3dxWare SmartUI now (by going to "Advanced Settings" in the window with SpaceMouse image).

pskowronskiTDx, 2024-05-16:
@andreymrt Thank you for the suggestion. We will update these tabs.
@chennes What is your preferred way of introducing the changes into Preferences window? Would you like us to remove these tabs on Windows or we should left one just for the information where to find the settings now?