FreeCAD/FreeCAD #13804: Using Part_EditAttachment fails when a Sketch is a member of a PD Body.

macdroid53, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The ellipsis from the Property view for Mapmode and Part_Edit Attachment produce essentially the same dialog.

When the selected sketch is a member of a PD Body the following error is displayed in an Error popup: Name 'basestring' is not defined

If a non-Body sketch is selected the dialog is displayed and it allows the user to select and attach the sketch to entities in a Body. But, does not add the sketch as a Body member.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37088 (Git)
Build type: Release
Branch: main
Hash: e4eb8598495fa6e3d470ea700eb928c42aaefafe
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * Pyramids-and-Polyhedrons
  * lattice2 1.0.0
  * Curves 0.6.35
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.4.20
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.13
  * Silk 0.1.5
  * Freecad-Built-in-themes-beta 1.2.2
  * kicadStepUpMod 11.1.3
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.20
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2024-05-30:
Just had another discussion about why this tool can't be used with PD Sketches.

Syres916, 2024-06-14:
@macdroid53 I'm tracking down the commit that caused this regression as I can't see any mention of it. This is a much bigger issue than just Part Design, it's affecting the use it was primarily designed for Part and Sketcher though I agree it shouldn't be chopped for Part Design either. I'll have a look at the commit and see what the thinking behind it was.

Syres916, 2024-06-14:
OK, getting somewhere, this https://github.com/FreeCAD/FreeCAD/pull/12885 appears to be the big change and the error I believe is generated around https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/AttachmentEditor/TaskAttachmentEditor.py#L59-L72 (which wasn't part of the PR) so I need to clarify how old the files need to be that should work because we can't have a situation where users have to rebuild 0.21.x and 0.20.x files just because of attached sketches, that seems too draconian IMHO.

Syres916, 2024-06-14:
@macdroid53 by all means give the PR change a try, you don't even have to recompile, just change the Python file in your _build folder/Mod/Part/AttachmentEditor_

macdroid53, 2024-06-14:
@Syres916 that seems to work!

closed 2024-06-17 by chennes