FreeCAD/FreeCAD #13805: Measure: Distance Factor and Mirror settings options need a tooltip

furgo16, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When going to Edit > Preferences > Measure > Appearance, the user can change the Measure tool settings. The function of at least two of them is not obvious, and would benefit from a tooltip (in addition to documentation):

- Distance factor
- Mirror

From a quick test, I've not noticed any difference in behavior when changing their setting, so I'm wondering if they are currently implemented.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git) Snap 1005
Build type: Release
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Core

### Anything else?

1. In fact, all settings on that dialog need a tooltip, but the two mentioned in this issue might be the most pressing. The rest of the options there are more or less self-explanatory.
2. As an aside, on the preferences page it would be logical to group the color-related settings: moving `Background color` to be after `Line color` might make more sense.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-03:
@hlorus maybe you can help what they should do. I also did not find any changes when placing with these.

hlorus, 2024-05-03:
These are actually not needed at all,  With UMF we have label dragging which makes the settings redundant. If i remember correctly the settings were ported over from Std_MeasureDistance where they would control the placement of the measurements.

maxwxyz, 2024-05-03:
should they be removed then or does this break stuff?

furgo16, 2024-05-03:
>

maxwxyz, 2024-05-03:
click and drag the label 😄 
It seems not possible when it is over geometry.

furgo16, 2024-05-03:
I had tried that with no success :-) My issue was that the distance between label and edge is reasonable for mm scale models, but it appears right on the edge for m scale models (Arch).

I'll see if I can upload a video. FWIW I'm using Gesture navigation, if that makes a difference.

hlorus, 2024-05-03:
They can be removed without breaking anything.

furgo16, 2024-05-03:
Reported the related issue as https://github.com/FreeCAD/FreeCAD/issues/13808

furgo16, 2024-05-03:
Indeed, their values seem to be defined, read, and then not used anywhere else:

https://github.com/FreeCAD/FreeCAD/blob/ede724366ea0cbb153301f7c43e4b597393a94f2/src/Mod/Measure/App/Preferences.cpp#L67-L70

https://github.com/FreeCAD/FreeCAD/blob/ede724366ea0cbb153301f7c43e4b597393a94f2/src/Mod/Measure/App/Preferences.cpp#L77-L80

PR #13811 (2024-05-13): [Measure] Remove unused properties from measure preferences
Remove leftover "Mirror" and "DistanceFactor" properties from measure preference dialog.

Closes #13805
closed 2024-05-13 by chennes