FreeCAD/FreeCAD #13808: Measurement: label distance needs to be dynamic depending on model scale

furgo16, 2024-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The distance between the measured element and the measurement label appears to be fixed to ca. 15-20mm: i.e. it does not adapt to the scale of the model. As such, the default works well for mm-scale models (Part, Part Design, etc.), but the label appears right next to the measured element for m-scale models (Arch).

When the cube's edges are 10 mm the label appears at a reasonable distance to the edges or faces:
[image]

When the cube's edges are 1000 mm the label appears stuck to the edges or faces (it's not, it just appears so as the element to label distance is small in comparison to the model):
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git) Snap 1005
Build type: Release
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-03:
the labels should be able to be dragged closer to the model. Each time I try to move it closer, it goes further away.

furgo16, 2024-06-26:
Closing this as fixed by #13848 . Note: for those using the Gesture navigation mode, labels can be dragged by holding the <kbd>Alt</kbd> key and then dragging with the mouse's left button.

maxwxyz, 2024-07-26:
The measurement is placed way off and cannot be moved closer to the model. It seems to be dependent of the model scale. This is the BIM example file:

https://github.com/user-attachments/assets/df8bd6fb-ae48-43b0-adfb-3a98e0c64fc1

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

maxwxyz, 2024-07-26:
I'll label this as blocker as this is a regression from the old measurement tool where the result is displayed at the location of the measurement.

hlorus, 2024-08-13:
> The measurement is placed way off and cannot be moved closer to the model. It seems to be dependent of the model scale. This is the BIM example file:
> 
>  measurement-placement.mov 
> ```
> OS: macOS 13.6.7
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.38314 (Git)
> Build type: Release
> Branch: main
> Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: C/Default (C)
> Installed mods: 
>   * freecad.gears 1.2.0
>   * OpenTheme 2024.7.3
>   * CfdOF 1.27.4
>   * Defeaturing 1.2.2
>   * Manipulator 1.5.7
>   * fasteners 0.5.25
>   * CurvedShapes 1.0.9
>   * sheetmetal 0.4.23
>   * Curves 0.6.42
> ```

I'm failing to find this example file, could you give me the link or send the file over discord?

maxwxyz, 2024-08-13:
It's right on the start page Help > Start under the example files at the bottom:
[image]

PR #15907 (2024-08-19): MeasureGui: Fix inital label placement
This solves the problem described in #13808 where the initial placement of simple label measurements was returned globally instead of relative to the root position. Also this PR adds a change to trigger the initial placement explicitly as the attach method (where it is currently triggered) is not the right place.

Fixes #13808
closed 2024-08-19 by WandererFan