FreeCAD/FreeCAD #13813: Arch Wall does not provide inner area

furgo16, 2024-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

Given a base wall of 3 m (H) x 3 m (W) x 0,4 m (T) as an example for the discussion. Defining **standalone walls** as those with a Draft Line `Base`, and **closed walls** as those with a `Draft Rectangle` `Base` –also for this discussion only.

[Arch_Wall](https://wiki.freecad.org/Arch_Wall) provides the `VerticalArea` property, which is inherited from [Arch_Component](https://wiki.freecad.org/Arch_Component) (unless mistaken, since it's not documented there). The vertical area returns the sum of the areas of all vertical wall faces.
- For standalone walls: the aggregated area of front, back, left, right vertical faces
- For closed walls: the aggregated area of outer front, back, left, right vertical faces **and** their inner counterparts

It also provides the `Area` property. It reportedly returns the area of the whole wall 
- For standalone walls: the area of the (front?) face. I.e. for our standalone wall, it returns 9m². Neither 18m² (front and back faces), nor 22.80 m² (all faces). In that regard, it does not work as documented. For the sake of discussion, we could call this `OuterArea`.
- For closed walls (e.g. with a rectangle baseline): the aggregated area of outer front, back, left, right vertical faces **without** their inner counterparts. For our closed wall, it returns 36 m². We could call this also `OuterArea`.

For a closed wall, there is no direct way to read what we could call `InnerArea`: effectively the aggregate of areas from the vertical faces inside the rectangle. For a standalone wall, this can be inferred from `OuterArea`, as they will be the same.

Given a use case of calculating the required area of insulation panels inside a house, it would be useful for Arch_Wall to provide `InnerArea` as well.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git) Snap 1005
Build type: Release
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-03:
The wall `Area` should actually be a simple calculation: Length x Height

Now indeed there is no way to get the inner wall area. I'm not sure how to do that, actually...

paullee0, 2024-11-22:
Just read this.  The 'Inner Area' is the length of the 'inner edges' x height, which can be calculated.

A few video posted on Youtube earlier back in 202 showcasting how the 'inner' floor area could be calculated automatically.
(e.g. https://www.youtube.com/watch?v=4NzBGluRwIA
https://www.youtube.com/watch?v=1XR1I9WG75Y)

https://github.com/user-attachments/assets/3b2c5b85-9e3a-4d44-8db5-a7788dd38216

https://github.com/user-attachments/assets/ef6fa2db-3dff-4337-98e2-57f7cbb365ae

furgo16, 2025-03-28:
A workaround is to create a dynamic property (e.g. `InnerArea`) with the property editor, as an expression. The expression can simply calculate the sum of the areas of all inner faces: e.g. 

```
.Shape.Face13.Area + .Shape.Face12.Area + .Shape.Face11.Area +.Shape.Face10.Area
```