FreeCAD/FreeCAD #13814: Start: Wrong language selected in Welcome screen

wwmayer, 2024-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

* Start FreeCAD with an existing user.cfg file.
* Open the Start page
* The pre-selected language is wrong

In the my preferences German is the selected language but the "Welcome to FreeCAD" shows a completely different language.
When opening the combo box then the languages are in a random order instead of sorted alphabetically. 

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git)
Build type: debug
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Reqrefusion, 2024-05-03:
The same. And as someone who hates Combobox, it's annoying to have to use it again. Generally, it chooses the next language from the system language.

chennes, 2024-05-04:
The bad selection is an off-by-one error in `index`. And I missed copying the sort code from the preferences dialog. Both corrected by #13823.

PR #13823 (2024-05-06): Start: Fix selection and ordering of language
Fixes #13814.