FreeCAD/FreeCAD #13816: Extruded Draft Facebinders should be able to close their corners

furgo16, 2024-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The [Draft_Facebinder](https://wiki.freecad.org/Draft_Facebinder) documentation states: 

> It can be used to create an extrusion from a collection of faces. This extrusion can for example represent a wall finish in architectural design. 

I've found out that not to be the case: when extruded using the `Extrusion` property, the resulting Facebinder does not join its corners. Expected: extruded facebinder has joined corners, or provides a property to join corners.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37085 (Git) Snap 1005
Build type: Release
Branch: main
Hash: c7c07f28c639e82a6d2c58f4aaef5b42cddcc83c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Draft

### Anything else?

- File to reproduce: https://forum.freecad.org/download/file.php?id=262640

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-06:
It's indeed the case. Faces are extruded individually. Of course this is not good, and faces should be united first then extruded as a whole.

I'm honestly not sure how to approach this, however. There are different extrusion directions, and a lot of corner cases... Is there some other tool of FreeCAD that would do a correct extrusion, that we could reuse here?

Roy-043, 2024-05-06:
You can use `sewShape` to create shells from faces. This is already done in the code, but before extruding the faces. However, Offsetting shells has some issues.
See: https://forum.freecad.org/viewtopic.php?t=87393

Roy-043, 2024-05-08:
Part 3D offset has the same issues. Internally it uses the same code as `makeOffsetShape`.

Roy-043, 2024-05-08:
Related forum topic: https://forum.freecad.org/viewtopic.php?t=87468
Part 3D offset also has an issue with a piramidal shell.

furgo16, 2024-05-08:
https://wiki.freecad.org/Part_CompJoinFeatures worked for the particular example in the forum, but it does not seem to work with Facebinder and the example file on this issue. Trying to use Connect on the Facebinder makes it disappear.

closed 2025-01-07 by yorikvanhavre

furgo16, 2025-01-08:
This works really well, thanks! For future reference, to make the extruded facebinder (`Facebinder003`) join in a miter, its `Sew` property needs to be set to `True`

PR #18901 (2025-01-07): Draft: Closed corners for extruded Facebinders
Fixes #13816.

The `makeOffsetShape` method that creates the extruded shape is quite picky. For example, it will work for a pyramidal shell (4 triangles) with a square floorplan, but not if the floorplan is slightly rectangular. To get closed corners the `Sew` property of the Facebinder must be set to `True`. If extruding does not work properly, the code will retry with `Sew` disabled.

There is also some code that tries to convert flat B-spline faces created between the main offset faces into planar faces. In some cases that code will fail (the r [... cut]
commit 6deb4245 (2025-01-07): Draft: Closed corners for extruded Facebinders (#18901)

* Draft: props_changed_placement_only should ignore material props

The new material related properties (Density, Volume and Mass) must be ignored by the `props_changed_placement_only` function.

Without this moving a Draft_Point will fail for example.

* Draft: Closed corners for extruded Facebinders

Fixes #13816.

The `makeOffsetShape` method that creates the extruded shape is quite picky. For example, it will work for a pyramidal shell (4 triangles) with a square floorplan, but not if the floorplan is slightly rectangular. To get clo [... cut]