FreeCAD/FreeCAD #13817: Measurement: Settings are available only after the first measure

pierreporte, 2024-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The UMF settings are only available after the first measurement has been done. Activating the tool is not enough. The problem is known for unloaded workbenches but this is in core and a very generic tool.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Anything else?

The problem is known for

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-15:
Can you check if it's still the case? In the newest Windows weekly, I can access the **Measure** preferences right after starting FreeCAD.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

pierreporte, 2024-06-15:
Using the same version, I can also see the settings right after start.

```
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.18
```

closed 2024-06-15 by pierreporte

FEA-eng, 2024-06-15:
> Using the same version, I can also see the settings right after start.

All right, thanks for checking so quickly. I was going to test it on my Ubuntu VM too.