FreeCAD/FreeCAD #13821: Materials tool crashes on arch workbench

furgo16, 2024-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Steps to reproduce

1. Create a new document
2. Switch to the Arch workbench
3. Click on the <kbd>Material Tools</kbd> (Arch_MaterialTools) icon
4. Material tools don't load, the user is presented with this error

```python-traceback
12:00:06  pyException: Traceback (most recent call last):
  File "/usr/share/freecad-daily/Mod/Arch/ArchMaterial.py", line 481, in setEdit
    self.taskd = _ArchMaterialTaskPanel(vobj.Object)
  File "/usr/share/freecad-daily/Mod/Arch/ArchMaterial.py", line 556, in __init__
    self.fillMaterialCombo()
  File "/usr/share/freecad-daily/Mod/Arch/ArchMaterial.py", line 699, in fillMaterialCombo
    for f in os.listdir(p):
<class 'FileNotFoundError'>: [Errno 2] No such file or directory: '/usr/share/freecad-daily//Mod/Material/StandardMaterial'
```

I can reproduce this both with the PPA and snap package builds. It looks like that (new?) folder needs to be added to the packaging sources to be distributed

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37096 (Git)
Build type: Release
Branch: main
Hash: af2b8b661f711f8512d63345e4ffe321e292ddbb
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * parts_library
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-04:
@davesrocketshop FYI

yorikvanhavre, 2024-05-04:
I'll attack this (compatibility with new Materials system) as soon as #13783 (BIM migratihn is done. |since the arch materials system is (currently) mostly self-contained, it shouldn't be much work to have it back working. Then we see about integrating things better

furgo16, 2024-05-05:
I've submitted a draft PR that addresses this: https://github.com/FreeCAD/FreeCAD/pull/13830

PR #13830 (2024-05-06): Arch: adapt materials to new Materials path scheme
Fixes #13821 by adapting the old material paths to the new scheme:

1. Update the materials filesystem path when reading them
2. Materials can now be in subfolders: traverse directories to read all materials

I'm aware of https://github.com/FreeCAD/FreeCAD/issues/13821#issuecomment-2094225362. I've addressed this as a quicker way to get back the Materials function before https://github.com/FreeCAD/FreeCAD/pull/13783 gets merged, which might take longer. The intention is certainly not to step on anyone's toes, so if this is already being worked [... cut]
closed 2024-05-06 by yorikvanhavre