FreeCAD/FreeCAD #13822: Reordering objects in the tree view resets their placement

alafr, 2024-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried using this new feature: https://github.com/FreeCAD/FreeCAD/pull/12293 but every object I drag and drop to reorder gets its placement reset to 0, 0, 0.

[image]

[test reordering.zip](https://github.com/FreeCAD/FreeCAD/files/15209533/test.reordering.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git)
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * ArchTextures
  * BIM
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * DynamicData 2.60.0
  * Trails 2022.1.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-04:
@PaddleStroke do you know this behavior?

alafr, 2024-05-05:
Closing because it's a duplicate of 13690

closed 2024-05-05 by alafr