FreeCAD/FreeCAD #13826: Camera doesn't always animate from the current orientation when focusing a sketch

jwueller, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When in Part Design, activating a sketch brings you into Sketcher. It also animates the camera to face the plane of the sketch. However, if you then rotate the camera _within_ Sketcher, and focus the sketch again, it instantly snaps to the original camera orientation you started out with in Part Design, and then does the camera orientation from that position, instead of where you were actually looking.

[freecad-camera-alignment-bug.webm](https://github.com/FreeCAD/FreeCAD/assets/388308/7110b319-08fe-45cb-8bf5-23233fce51f1)

I only noticed this in Sketcher so far, but this might affect other contexts too. #11234 is tangentially related, since doing camera alignment to arbitrary features probably has to deal with this at some point anyway.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-05:
So the issue is, that there is no smooth animation? Is this also the case when you right click in the active sketch and select view sketch?

jwueller, 2024-05-05:
No, the issue is that the animation sometimes starts from the wrong position, which suggests it's using stale data. The animation itself is perfectly smooth. Yes, this also happens if I re-view the sketch from the context menu.

maxwxyz, 2024-05-05:
@Rexbas you are the guy for this 😄

Rexbas, 2024-05-05:
Yes, I will look at it.

Rexbas, 2024-05-10:
I looked into it and currently, when you exit the sketch mode, the camera is restored to its orientation before you entered the sketch mode, without animations.

The problem is that when you double click on the sketch in the tree view then the sketch mode is exited and entered again. This means it restores the camera first, then it starts an animation to the sketch plane again. I do not know how to check if a sketch is refocused by double clicking in the tree view or if a sketch is closed. Thus, I have no way to prevent the restore code from running.

However, it is possible to disable the camera restore completely using the parameter editor (Tools -> Edit parameters...). Set `RestoreCamera` to false in `BaseApp/Preferences/Mod/Sketcher/General`. I personally would not mind if the default is false because I think the restore is disorienting. But others may disagree.

jwueller, 2024-05-10:
Curious, any idea why the sketch even closes when you (re)activate it in the tree? Or is the tree just hard-coded to always exit whichever mode the viewport is in and then enter the clicked one, without regard for whether it's already active?

Rexbas, 2024-05-10:
It indeed exits whatever is active first, regardless of what is active. However, this code is not specific to the tree view: https://github.com/FreeCAD/FreeCAD/blob/7d738f161a64b1fc154abed244603a89b3439617/src/Gui/Document.cpp#L296-L301

maxwxyz, 2024-05-10:
the entire double click in the tree is a mess..
https://github.com/FreeCAD/FreeCAD/issues/12839

luzpaz, 2025-07-12:
re-opening since it's not clear this issue is fixed.

FEA-eng, 2025-09-14:
@Rexbas Still relevant?

Rexbas, 2025-09-14:
Yes still relevant.

closed 2026-09-07 by maxwxyz