FreeCAD/FreeCAD #13827: On-View-Parameter inputs show at the wrong coordinates

jwueller, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems that depending on the attachment settings, the values sometimes refer to the wrong axis, which leads to very strange behavior.

[image]

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-05:
can you provide a demo file or video? It is unclear how to reproduce.

jwueller, 2024-05-05:
I'm still trying to isolate what causes this. I'll upload a minimal reproduction when I figure it out.

jwueller, 2024-05-05:
Here we go: [FreeCAD-On-View-Parameter-Coordinate-Bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15212738/FreeCAD-On-View-Parameter-Coordinate-Bug.FCStd.zip)

It only contains a few elements, and the sketch projection clearly differs from the one used to draw the parameter UI.

To me, this suggests that there is more than one way to get from viewport coordinates to world coordinates and back, which is probably a more fundamental issue, but this might be a good way to get to the bottom of it.

maxwxyz, 2024-05-05:
thank you, I can reproduce:
[image]

@PaddleStroke FYI, there seems to be an issue with the OVPs.

jwueller, 2024-05-05:
If you rotate the camera in sketch view, you can see that it's not even in the same plane as the sketch sometimes:
[image]

PR #13844 (2024-05-06): Sketcher: fix OVP placement issue #13827
Fixes #13827
closed 2024-05-06 by adrianinsaval