FreeCAD/FreeCAD #13828: AddonManager: Composite view resize issue

maxwxyz, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the composite view this is the default size the addon manager will open on my system. It is not possible to enlarge the right view to the left in this state. There seems plenty of space in the list, but I guess the top bar for sorting/searching/filtering restricts the size of the left column to be smaller. Maybe this could use the full width (marked in green):

[image]

When I want to read the content of the selected addon I have to increase the entire addon manager dialog window to the right. During increasing the right view stays the same size, so only the left side is increased. After resizing, I need to manually adjust the right view to increase its size to the left again.
Increased width state:
[image]

@chennes FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git)
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-05-05:
Because that area has to be reserved for the install/etc. buttons, I can't easily allow another widget to impinge: the two regions are completely separate at the code level.

maxwxyz, 2024-05-05:
Would there be a way to use another region on top of both views?

chennes, 2024-05-05:
Not reasonably -- in a recent refactor I separated the two things specifically to eliminate code duplication and enable the creation of Composite View. And as soon as you select an addon, you're going to lose that space anyway. So part of your report is fixable (it should be remembering your previous size), but using that "extra" space isn't feasible in part because it's only extra until you have an addon selected.

maxwxyz, 2024-05-05:
mh, then I have no idea for this. What about two rows, filter and sort in the second row?