FreeCAD/FreeCAD #13831: TechDraw: Snapping Dimensions: Angles/Diameter/Radii

maxwxyz, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following up on https://github.com/FreeCAD/FreeCAD/pull/13659

Snapping dimensions should also be possible for:
1. dimension value to midpoint of radius dimensions
2. dimension value to midpoint of diameter dimensions
3. dimension value to midpoint of angle dimension
4. dimension lines between arc dimensions
5. cascading distance between angle dimension lines
[image]

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git)
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Techdraw

### Anything else?

Related:
- https://github.com/FreeCAD/FreeCAD/issues/11636

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-29:
This is a good project for a new contributor if anyone wants to jump in. As snapping is already introduced, making new kind of snapping should be pretty straight forward. It all happens in `QGIDatumLabel::snapPosition`. Just needs to define the various cases and make them to work properly.

maxwxyz, 2026-11-07:
Part of this is in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/674092cbc48bc9c2c7edf72dfeb651092e6b91e7

Radius, diameter and angle labels can snap into place. For radius and diameter this only covers ISO oriented and ASME inlined, the referencing styles are not supported.