FreeCAD/FreeCAD #13834: FEM: Reorder filters in results pipeline with drag & drop

maxwxyz, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From: https://github.com/FreeCAD/FreeCAD/pull/13820#issuecomment-2094807173

Currently it is not possible in the FEM WB to reorder the filters in the result pipeline.
It should be implemented in the drag & drop methods.
It must be considered that the inputs could change, when he input of the current filter is based on the previous if the order is changed.

@marioalexis84 / @FEA-eng  FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git)
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-30:
@FEA-eng is this still an issue?

FEA-eng, 2025-06-04:
> [@FEA-eng](https://github.com/FEA-eng) is this still an issue?

It is now possible to drag and drop the filters within a pipeline or a branch.