FreeCAD/FreeCAD #13841: Expression engine internally points to the wrong open document

krushia, 2024-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After doing "Save As..." to a different file, the expression engine might point to the OLD file until FreeCAD is restarted. This is tricky to reproduce, but can lead to data corruption.

Partial example:
1. Create a new project
2. Create a sphere
3. Create a cube
4. Change Box.Label to the expression `<<Not>> + Sphere.Label`
5. Save the document
6. In the Tree View, right click > Expression Actions > Copy Selected Document. Note that in the copied text, the references will point to the unsaved document instead of the current document name, like `Unnamed#Box.ExpressionEngine`

At this point, I tried different combinations of renaming the saved document and opening it, and somehow managed to get changes in Sphere.Label to affect Box.Label, but I can't reproduce it reliably. It might be some weird race condition? In my case I ended up with two open tabs, one a "wrong-doc-bugtest" and the other a "wrong-doc-bugtest1" that were cross-referencing each other in memory.

When in the "failing" mode, both open documents showed:
`##@@ Label wrong_doc_bugtest1#Box.ExpressionEngine (NotSpherex)
##@@
<<Not>> + Sphere.Label`

I ran into this in a larger project, where I Opened the file "foo-mk3.FCStd" and after doing Save As foo-mk4, foo-mk5, foo-mk6... I found out when re-opening foo-mk3 that my still open foo-mk6 was suddenly using old values from a spreadsheet in foo-mk3. Am I making any sense?

Sorry, I wish I could reproduce it reliably :(

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * A2plus 0.4.64a (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * Nodes 0.1.36 (Disabled)
  * MnesarcoUtils 0.2.5 (Disabled)
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-07:
I'm not sure it will be possible to make an example file. There seem to be 2 separate problems here. The first is an annoyance, and _may_ have resulted in user error on my part.

> The Document.Name in FreeCAD is not restricted to a Type[Number] like other objects (Box001, Sphere003, etc). Instead it is AnyString[Number], with AnyString being "Unnamed" (or translation?) for a new document, and filename without extension when opening an existing file. If AnyString matches an existing name, FreeCAD tries to change the name by removing any trailing digits, then adding a digit. Problems arise when AnyString has digits at the end already.

1. Following original "Partial example" save the file in step 5 as foo_1.FCStd
2. Completely close FreeCAD
4. Do some stuff (optional), save it as foo_2.FCStd
5. Open foo_1.FCStd
6. At this point, you now have a mess

```
>>> App.getDocument('foo_1').FileName.rsplit('/')[-1]
'foo_2.FCStd'
>>> App.getDocument('foo_2').FileName.rsplit('/')[-1]
'foo_1.FCStd'
```

My suggestion here would be to name every Document.Name in FreeCAD with the format 'Document'[number] to avoid dummys (me) from treating it as reference to a filename.

As for the second issue, it here is a summary:

1. Document.Label is checked for uniqueness when first opened, but not afterwards.
2. Document.OldLabel is never checked for uniqueness.
3. When Document.Label is changed, it is possible for Document.OldLabel to be the same as another Document.Label
4. The expression engine will perform erroneous replacements in such a case

Reproducing this is difficult, as it has to do with timing (I think). Still working on that.

maxwxyz, 2024-10-30:
Is this the same or solved with a fix for https://github.com/FreeCAD/FreeCAD/issues/12845

closed 2025-07-09 by github-actions[bot]

FEA-eng, 2025-07-17:
This is still relevant.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```