FreeCAD/FreeCAD #13843: Multiple cuts on spheres cause issues.

Hale9000, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following code demonstrates the problem:

def demonstrate_cut_bug():
    """
    Code that shows an issue when Angle1 is set to 0 or a negative number. Set to 0.01 or greater, and it works as expected
    """
    doc = App.ActiveDocument
    radius = 20
    part = doc.addObject("Part::Sphere", f"Top Sphere")
    part.Radius = radius
    part.Angle1 = -0.01
    bp = BOPFeatures(doc)
    for angle in [0, 60, 120, 180, 240, 300]:
        # Cut with spheres instead because geometry is more simple
        sp = doc.addObject("Part::Sphere", f"Cut Sphere {angle}")
        x = round(radius * math.sin(math.radians(angle)), 4)
        y = round(radius * math.cos(math.radians(angle)), 4)
        sp.Placement = Placement(Vector(x, y, 0), Rotation(App.Vector(0, 1, 0), 0))
        sp.Radius = radius
        part = bp.make_cut([part.Name, sp.Name])

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

Part

### Anything else?

For some reason it happens when it is sliced by the 5th sphere. Changing the order of the slicing can fix or modify the bug. However, when the object is later sliced the problem reappears.

Expected:
[image]

Bug:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-06:
not sure if this is related to OCC?

Hale9000, 2024-05-10:
Here are the imports:

import FreeCAD as App
from FreeCAD import Vector, Placement, Rotation
from BOPTools.BOPFeatures import BOPFeatures

import math

Hale9000, 2024-05-14:
**You must have the Part workbench loaded in FreeCAD.**

Here is another demonstration with even stranger results:

def demonstrate_cut_bug2():
    """
    Code that shows an issue when offset set to 0
    """
    doc = App.ActiveDocument
    radius = 20
    offset = 0
    part = doc.addObject("Part::Sphere", f"Top Sphere")
    part.Radius = radius
    # part.Angle1 = 0
    part.Angle3 = 360
    part.Placement.Rotation = App.Rotation(App.Vector(1, 0, 0), 90)
    # return
    bp = BOPFeatures(doc)
    for angle in [0, 60, 120, 180, 240, 300]:
        # Cut with spheres instead because geometry is more simple
        sp = doc.addObject("Part::Sphere", f"Cut Sphere {angle}")
        x = round(radius * math.sin(math.radians(angle)), 4)
        y = round(radius * math.cos(math.radians(angle)), 4)
        sp.Placement = Placement(Vector(x, y, offset), Rotation(App.Vector(0, 1, 0), 0))
        sp.Radius = radius
        part = bp.make_cut([part.Name, sp.Name])

[image]

With offset set to 1 you get the expected results:

[image]

maxwxyz, 2024-11-01:
@AIRCAP FYI :)

luzpaz, 2024-11-01:
@Hale9000 please add the imports to the existing code so testers can easily cut/paste to test it

closed 2025-09-10 by github-actions[bot]

maxwxyz, 2025-09-10:
@luzpaz why are you reopening the issue, are you able to reproduce the issue?
If not, there is no benefit from the stale action.

I just tried it on the new weekly and cannot reproduce the issue: not with Angle 0, -0,01 or 0,01:

[image]

Closing for now. 
IMO we should only reopen if there is a response or the issue is reproduced on the latest weekly.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * AddonManager 2025.8.16
  * Cables 0.2.1
  * CurvedShapes 1.0.13
  * Defeaturing 1.2.6
  * dodo 1.0.1
  * freecad.gears 1.3.0
  * Gridfinity 0.11.9
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.2
  * Rocket 4.1.1
  * sheetmetal 0.7.24
  * Silk 0.2.0
  * Telemetry 1.0.2

```