FreeCAD/FreeCAD #13846: Start: files created during current FreeCAD session are not added

Roy-043, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The old Start page would also show files created during the current FreeCAD session. This is no longer the case with the new implementation.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git)
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-06-06:
I noticed this recently as well.  Would be good to see it resolved before 1.0 is released.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

JDMils, 2024-06-23:
I concur that this issue still exists in build 37573. As a work-around, recent files can still be accessed via Files \ Open Recent.

Beep6581, 2024-11-24:
Consider renaming this issue's title to make it easier to find. Use the keyword "Recent Files".

Still valid in 45134cef165ecc11aca69f81673e7b69421cebf1

The issue: when you open a file in FreeCAD, it gets added to the list of recent files, but the "Recent Files" panel on the "Start" page is not updated/reloaded to reflect this. You can only see the file once you restart FreeCAD.

furgo16, 2024-11-24:
Issue #15110 is not a duplicate. It describes a different behavior and is specific about images.

Beep6581, 2025-07-24:
Fix confirmed.