FreeCAD/FreeCAD #13847: Mesh display mode should be Flat Lines by default

FEA-eng, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, meshes (created in the Mesh wb or imported from STL) have the Display Mode set to Shaded by default. IMO Flat Lines would make more sense as one usually wants to see the division into triangles.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git)
Build type: Release
Branch: main
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-11-14:
@maxwxyz should the DWG weigh-in on this ?

maxwxyz, 2025-11-14:
I think it is just that the flat lines draw style in the mesh view provider does not highlight the triangulated edges currently. It should not change default draw style but the flat lines should be supported in mesh.