FreeCAD/FreeCAD #13849: Colored coincidences don't work on perpendicular or on ends of B-splines

chrisb-github, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1) Poin-to-point perpendicular constraints are coincident and should have the same color as the latter.
[image]

2) Coincidences, Point-to-point tangents and perpendicular constraints at the ends of B-splines should have the same color as they have e.g. for arcs.
[image]

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git)
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.15
  * ExplodedAssembly
  * Curves 0.6.29
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-06:
@PaddleStroke FYI

PaddleStroke, 2024-05-06:
There's another issue opened on this.
- perpendicular yes.
- bspline : in fact the internal geo alignement points are drawn on top of the coincident points. As it's supposed to be, like if any point is not coincident its drawn on top. But yes in this case it's bad. An exception should be done for bspline end internal points