FreeCAD/FreeCAD #13850: Assembly: Min. limit of joint issues

maxwxyz, 2024-05-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
Minimum limit of joints (distance and angles)
Can be larger than the max limit. If min limit is e.g. 10 and max. limit is 0 it still can move freely.

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DrD4ffy, 2024-05-07:
The slider limits confuse me. To move the upper cylinder up to 10 mm in positive direction along the z-axis I have to set limits from 0 to -10 mm. Have I misunderstood something, or is this another bug?
[image]

PaddleStroke, 2024-05-07:
The thing is that there are 2 ways to look at your drawing. Either Cylinder1 is 10mm in front of cylinder2, or cylinder2 is 10 mm behind cylinder1.
In other terms the order of the joint selection matter. If you selected the other way, then it would have been 0 to 10.

I need to think about how to make this more intuitive.

DrD4ffy, 2024-05-07:
OK, understood. I was misled by the fact that the base cylinder is grounded and only the upper one moves - definitely in positive direction :-) Perhaps the LCS that is in some sort the origin of the joint/movement could be visualized differently? And perhaps the order could be changed if one part is grounded?

maxwxyz, 2024-05-07:
@PaddleStroke When both limits are 0, is this regarded as a limit at all? So is it possible to just set a min limit or just set a max limit and don't limit in the other direction? If so, maybe the second option seems good. If the user inputs first min to 10 and max is still 0, it could be considered as no limit. Otherwise maybe do the evaluation at creation/when user clicks ok. If not possible the first option as workaround. 
Is it possible to insert expressions in limit/offset fields? If so this needs to be considered when you want to change min/max if they flip, even when task view is not open.

closed 2024-06-06 by maxwxyz