FreeCAD/FreeCAD #13852: Sketcher dimensional constraint text rendered backwards

krushia, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Seemingly at random, opening a Sketch will show all the constraints backwards (as if seen in a mirror or from behind).
[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-07:
please provide steps to reproduce or a demo file.

krushia, 2024-05-07:
[krushia-clampstrip-mkJ-22dev.zip](https://github.com/FreeCAD/FreeCAD/files/15235853/krushia-clampstrip-mkJ-22dev.zip)

Double-click a sketch in the tree. Then double-click a different sketch... keep doing this and eventually the problem arises. I just tried it and it happened on the second opened sketch. Sometimes it takes multiple tries.

maxwxyz, 2024-05-07:
I can reproduce.
It is correct on opening any of the sketches for the fist time. After opening another one it is flipped.
It does not flip when I actively close the previously opened sketch with the close button in the task dialog. So when I have a sketch open and double click on another sketch, the new one is flipped. 
@PaddleStroke FYI

When looking from the other side I dont see dimensions at all (https://github.com/FreeCAD/FreeCAD/issues/13557) is it the same on your side?

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/60a47bf7-6286-4971-a2ef-eb2c4e320e36

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.3
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

krushia, 2024-05-07:
I can't reproduce #13557. In this case, when viewing from the back side the dimensions are still shown and backwards.

maxwxyz, 2024-06-27:
It shoukd be fixed now, @krushia please report back

krushia, 2024-06-29:
Bug still exists, I tried these two versions:

Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123

Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b

maxwxyz, 2024-06-29:
@FlachyJoe any idea about this?

krushia, 2024-06-29:
This is probably happening because sketches are in a `Part::Mirroring`. Why they sometimes show in reverse and sometimes not is a mystery though. Seems to depend on the order they are opened for editing.

krushia, 2024-07-01:
Got another example, this time no `Part::Mirroring` (or any other mirroring). Double-clicking `CenterPillarWSnapYZSketch` comes up with reversed dimensions every time. This sketch was showing just fine a few minutes ago...

Also attaching screenshot of history (from Undo button, from the document this thing was created in) since sketch creation. Sadly my console is full of junk from other open documents.

[another-bug-13852-example.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16059788/another-bug-13852-example.FCStd.delete.this.extension.zip)

[image]

[image]

If someone can tell me how to print undo history in the terminal I can do that.

luzpaz, 2024-07-24:
@FlachyJoe do you have a moment ?

krushia, 2024-08-01:
I now have another issue that may be related (evolution of the same cursed project). Now zooming in Sketcher is messed up. Open the file, find a sketch (the one in screenshot is `DualCantileverClipDualClipTemplate`). Now zoom in. The dimension text somehow scales down to nothingness (I thought it never scaled?!?) and the geometry gets "stuck" around where the text vanishes and stops scaling (in screenshot I'm scaled down to nanometers but the display is showing geometry about 15 mm high).

Interestingly, flipping the view around makes the zoom buggy in the other direction I think. IDK I'm very confused...

[krushia-clipstrip-202407311933.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16448601/krushia-clipstrip-202407311933.FCStd.delete.this.extension.zip)

[image]

krushia, 2024-08-01:
While trying to simplify that file, I realized that `Origin002` is tied to both `Part001` and `Body` (which is nested a few fusions deep in `Part001'). This could be the problem.

maxwxyz, 2024-08-04:
any news on this issue?

krushia, 2024-08-11:
There is an old report that seems to be the same thing: #6040

I tried two of the files from the forums referenced in that report, and they both have the issue. I suppose that makes this a regression since 19.0? Those files are attached here.

From [toorxBX300_holder.fcstd.delete.this.extension.zip](https://github.com/user-attachments/files/16576282/toorxBX300_holder.fcstd.delete.this.extension.zip) the "Hole Base" sketch:
[image]

From [sample_to_forum.fcstd.delete.this.extension.zip](https://github.com/user-attachments/files/16576284/sample_to_forum.fcstd.delete.this.extension.zip) the "sketch" sketch:
[image]

luzpaz, 2024-09-10:
There are some other mirroring bugs...need to find if they are dupes

PaddleStroke, 2025-12-30:
Finally I found what's wrong in those files. The sketches are actually not in the body.

https://github.com/user-attachments/assets/4212decc-1319-49a8-9068-d7b1ca8bf9cb

PaddleStroke, 2025-12-30:
So this ticket needs to be re-opened, 
https://github.com/FreeCAD/FreeCAD/pull/25478 and https://github.com/FreeCAD/FreeCAD/pull/26033 need to be reversed.
https://github.com/FreeCAD/FreeCAD/issues/26481 will be closed by the reversal.
https://github.com/FreeCAD/FreeCAD/pull/26531 can be closed.

Now the question is what to do about this ticket here. Anyone know how user end up with such documents?

PaddleStroke, 2025-12-30:
Note, I have analysed the file [toorxBX300_holder.fcstd.delete.this.extension.zip](https://github.com/user-attachments/files/16576282/toorxBX300_holder.fcstd.delete.this.extension.zip).

Which may be a different bug than the other files in this ticket...
However I tested [krushia-clampstrip-mkJ-22dev.zip](https://github.com/FreeCAD/FreeCAD/files/15235853/krushia-clampstrip-mkJ-22dev.zip) and cannot replicate the issue of this file.

PaddleStroke, 2025-12-30:
Yes let's @ipatch confirm first. Since there are multiple documents provided maybe I missed something

luzpaz, 2025-12-30:
> So this ticket needs to be re-opened,

Reopening ticket

PaddleStroke, 2025-12-31:
Finally I got it!

in unsetEdit we do : 
```
void ViewProviderSketch::unsetEdit(int ModNum)
{
    ...
    // clear the selection and set the new/edited sketch(convenience)
    Gui::Selection().clearSelection();
    Gui::Selection().addSelection(editDocName.c_str(), editObjName.c_str(), editSubName.c_str());
```

This convenience is what is causing the bug. When you double click on a new sketch, it calls resetIfEditing which call the old sketch unsetEdit. This modify the selection by clearing the new sketch and adding the old sketch. Then ParentFinder runs, and obviously cannot find anything since the selection has been changed.

closed 2026-01-05 by chennes