FreeCAD/FreeCAD #13853: Assembly: Dragger rendered to large / DPR respect

maxwxyz, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new dragger is visible when selecting an object in the tree view of an active assembly.
IMO it is rendered way too large (compare with mouse cursor).
Is there a setting to reduce it's size or disable it?
[image]

@PaddleStroke FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * fasteners 0.5.20
  * OpenTheme 2024.4.20
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2024-05-08:
@maxwxyz I can quickly tackle this. Would making it smaller suffice or should there also be a parameter or preference?

maxwxyz, 2024-05-08:
@obelisk79 what is your opinion on this?

obelisk79, 2024-05-08:
I think a parameter would suffice

maxwxyz, 2024-07-12:
@qewer33 any news on this?

PR #15401 (2024-07-15): Gui: Reduce Transform Dragger size
The size can now be configured using `User parameter:BaseApp/Preferences/View/DraggerScale` parameter.

In code value of this parameter is exposed as convenience function `SoFCCSysDragger::getScaleFromUserParameters` which adds dependency from `Gui` to `Application.h`. I'm not sure if this is ok - @wwmayer if it is not please point to a better place for such function.

Before:
[image]

After:
[image]

Fixes: #13853
commit 284c42dc (2024-07-15): Gui: Reduce Transform Dragger size

Fixes: #13853
closed 2024-07-15 by chennes