FreeCAD/FreeCAD #13856: [Problem] Issues with compiling FreeCAD from source on Linux（centos7）

sigurHeBin, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hey!
I am trying to compile FreeCAD from source for Linux. I am using the following script to do that:
```
[root@iZn11er29gnfasZ build]# cmake -DQT_RCC_EXECUTABLE=$(which rcc) ../
CMake Deprecation Warning at CMakeLists.txt:8 (cmake_policy):
  The OLD behavior for policy CMP0072 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-7/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-7/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiler: GNU, version: 7.3.1
-- Looking for GL/gl.h
-- Looking for GL/gl.h - found
-- Looking for C++ include istream
-- Looking for C++ include istream - found
-- Looking for C++ include ostream
-- Looking for C++ include ostream - found
-- Looking for C++ include fstream
-- Looking for C++ include fstream - found
-- Looking for C++ include sstream
-- Looking for C++ include sstream - found
-- Looking for C++ include ios
-- Looking for C++ include ios - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Looking for C++ include iomanip
-- Looking for C++ include iomanip - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Check for STD namespace
-- Check for STD namespace - found
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib64
-- cmake: 3.28.4
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.5") found components: doxygen missing components: dot
-- Found Python3: /usr/local/bin/python3.8 (found version "3.8.17") found components: Interpreter Development Development.Module Development.Embe                               d
-- Found XercesC: /usr/lib64/libxerces-c.so (found version "3.1.1")
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.7")
-- PyCXX found:
--   Headers:  /app/software/FreeCAD/src
--   Sources:  /app/software/FreeCAD/src/CXX
--   Version:  6.2.8
CMake Warning (dev) at /usr/local/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (OCC) does
  not match the name of the calling package (OpenCasCade).  This can lead to
[... 137 lines cut ...]
there is a error：CMake Error at cMake/FindPySide2Tools.cmake:78 (message):
  Qt rcc is required for generating Resources/AddonManager.qrc
Call Stack (most recent call first):
  src/Mod/AddonManager/CMakeLists.txt:2 (PYSIDE_WRAP_RC)

But I have successfully installed QT RCC and have specified the path in cmake：
```
[root@iZn11er29gnfasZ build]# which rcc
/usr/local/Qt-5.12.12/bin/rcc

``` 

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/tree/0.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-05-07 by sigurHeBin

zhaosky, 2024-07-05:
how your reslove this problem

JackyhuangUp, 2024-07-10:
Is there a solution now?

sigurHeBin, 2024-07-11:
> how your reslove this problem

version issue,i change the FreeCAD version to 0.21.2 and install a specific version of qt.