FreeCAD/FreeCAD #13858: [Problem] V0.22.0dev(37213) - [Sketcher]  Can't select origin or origin axis

7th-fret, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With Version: 0.22.0dev 37213 i can't select origin or origin axis in sketcher.
With previous build (37108) the problem doesn't exist.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * InventorLoader 1.4.0
  * Curves 0.6.35
  * freecad.gears 1.2.0
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
  * fasteners 0.5.20
  * OpticsWorkbench 1.0.17
  * Render 2024.1.23
  * OpenDark 2024.3.13
  * A2plus 0.4.64a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-07:
Let me clarify, the axis and origin don't change colour when selected which is still a regression but they are selectable.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Design456 0.0.1
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * fasteners 0.5.20
  * freecad.gears 1.1.0

```

DrD4ffy, 2024-05-07:
I cannot select origin, axis or external geometry lines when I try to apply constrain Coincident/PointOnObject (e.g. Concident of one endpoint of a line and origin or PointOnObject of one endpoint on axis or external geometry)
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git)
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.35
  * fasteners 0.5.20
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.2.0 (Disabled)
  * Help 1.0.3
  * IconThemes
  * lattice2 1.0.0
  * Lithophane
  * OpenTheme 2024.5.3
  * QuickMeasure 2022.10.28
  * Render 2024.1.23
  * sheetmetal 0.4.13 (Disabled)
```

Syres916, 2024-05-07:
OK, this is even more bizarre because the selection output in the Python Console is correct in my testing and dimensions from the axis work for me but I can also confirm @DrD4ffy observations.

The root cause appears to be https://github.com/FreeCAD/FreeCAD/commit/d7ade12dc320ff599e2cb57bece3db8eaa83cfe9 as when I only backout the _SketchObject.cpp_ changes the axes are now changed in colour and I can constrain the end of a line to the origin. @bgbsww could you please have a look, TIA.

bgbsww, 2024-05-07:
I'm sure there is, but I'm not sure what.  Let me dig in on that.  I assume you were selecting the vertical axis, so the question is why that's getting in getSubTopoShape.  That will need a traceback which I'll just generate in a debug session. 
 
Thanks!

Syres916, 2024-05-07:
@bgbsww if you're struggling to replicate the error when selecting the axis, don't waste any further time on it, I've just reset my config files and recompiled to Git revision 37225 with the line in _SketchObject.cpp_ uncommented and the errors are no longer appearing.

Andrei-Pozolotin, 2024-05-07:
@chennes:

same(?) problem is still there:

[issue_13858.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15238334/issue_13858.FCStd.zip)

clicking "origin" in the attached sketch produces:

```cpp
11:18:35  <Exception> TopoShapeExpansion.cpp(2031): Invalid shape name RootPoint
11:18:35  6.93705 Selection.cpp(155): Unhandled Base::Exception caught in selection observer: 
```

```
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37227 (Git)
Build type: Release
Branch: main
Hash: c1bc3b6e9a404ce68e4480711aba1972a12241f3
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

bgbsww, 2024-05-07:
Thanks Andre-Pozolotin and Sysres916.  I think these are unrelated issues.  This issue ( can't select ) is solved by the quick patch.  The example file does generate the Exceptions, so that's great - I can look from there.