FreeCAD/FreeCAD #13861: GUI: Cannot change the font size of the Python console by CTRL+mouse scrolling, like the Report view

yorikvanhavre, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Report view, it is possible to quickly change the font size by pressing CTRL + using the mouse wheel. The same is not possible on the Python console. This should be good to have.

Apparently both are using a QTextEdit, I didn't find any code in FreeCAD that manages that mouse scrolling, so I suppose it's built-in... Couldn't find why it applies to one and not the other, though...

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 +10 (Git)
Build type: Unknown
Branch: BIM-migration
Hash: f3c619086ae2180bcc3f3dd765ffe8b1ecb31ba8
Python 3.11.9, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * dxf-library
  * WebTools 1.0.0
  * parts_library
  * Ondsel-Lens 2024.1.22.01
  * OpenDark 2023.12.17
  * Freecad-Built-in-themes-beta 1.2.2
  * Reinforcement
  * Dracula 0.0.5
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-06:
This is because wheel zoom is only allowed if text isn't editable. We therefore have to reimplement that part.

benj5378, 2024-10-06:
I didn't see #15891 was already made... but perhaps it's better to place soltuion on TextEdit level than PythonConsole? @xtemp09

xtemp09, 2024-10-06:
@benj5378, I don't mind, it is for developers to decide.

 The case is about Python console, not about something else.  Potentially, your commit can break something else, so, I would first make it turnoffable.

benj5378, 2024-10-06:
@xtemp09, thanks for your input.

TextEdit is only inherited by `PythonConsole` and `PythonEditor`. So adding the fix to `TextEdit` will also make the macro editor zoomable, so I think that is the best way to go. As it's only `PythonConsole` and `PythonEditor`, I don't find i necessary to make it turnoffable.

If I close my PR, can you update your PR by moving the fix to `TextEdit`?

xtemp09, 2024-10-07:
@benj5378, I don't mind if developers close my PR in favor yours.

Though, I'm curious why @maxwxyz tagged my PR with `Post 1.0`, while your with `1.0`. Your commit potentially can have more side effects or just be undesirable. It must be discussed and more carefully tested; I'm not sure how it behaves with all this autocompletion and highlight and if it actually is desired.

maxwxyz, 2024-10-07:
@xtemp09 sorry, I saw it as a new feature then, not as fix for accessibility issues. 
I've set your PR also to the 1.0 milestone. Dev's should decide between #15891 and #17053

PR #17053 (2024-10-07): [Gui] TextEdit, zoom on Ctrl + wheel
Fixes #13861
commit 13d00f98 (2024-10-07): [Gui] TextEdit, zoom on Ctrl + wheel

Fixes #13861
commit c4e71656 (2024-10-14): [Gui] TextEdit, zoom on Ctrl + wheel

Fixes #13861
closed 2024-10-07 by yorikvanhavre