FreeCAD/FreeCAD #13863: Spreadsheet loses focus when opened from detached Tree View

krushia, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the Tree View panel is detached from the main window, opening a spreadsheet from it will "flash" the spreadsheet for a moment, then it will lose focus to whichever tab was selected previously.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Happens in stable 21.2 as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-26:
@krushia can you provide a screencast, I can't seem to reproduce on  
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

xtemp09, 2024-12-07:
Is this issue still relevant?

tarman3, 2024-12-07:
Yes

https://github.com/user-attachments/assets/392bf14d-68b3-48ac-81e7-4bae0cd36efe

<details>
  <summary>FreeCAD 1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

FEA-eng, 2025-09-14:
I can't reproduce it on Windows.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-08-16:
@tarman3 Please confirm if it's still relevant.