FreeCAD/FreeCAD #13864: [Feature Request] Automatic Alias creation tool

NXTler, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Greetings,
When having multiple variables, it's very likely that you will arrange them in a list like this and change the Alias' accordingly for usability:
[image]
This is really annoying, since you will have to type the name twice and in different forms (Gehäuse Länge > Gehaese_Laenge). It would be great to have a tool to do this automatically.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.4.2
  * A2plus 0.4.63
  * QuickMeasure 2022.10.28
  * Curves 0.6.23
  * CfdOF 1.25.2
  * IconThemes
  * fasteners 0.5.12
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-05-07:
checkout EasyAlias macro

NXTler, 2025-11-21:
This is essentially solved with the macro, but it could be turned into a default function.