FreeCAD/FreeCAD #13871: LGPL license violation

Zolko-123, 2024-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD comes with the LGPL license, which requests that the source-code be redistributed along with the binary distribution. Unfortunately, this is not satisfied with FreeCAD's 0.21.1 version release :

https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.1

It is possible there to download binary executables for various platforms, and there is an archive for the corresponding source code, but :

**1.) the code doesn't actually compile, doesn't even configure**

It gives the error: 

`-- Configuring incomplete, errors occurred!`

When looking into the log file, there is only 1 error and it reads:

```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake cmTC_0d978/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_0d978.dir/build.make CMakeFiles/cmTC_0d978.dir/build
gmake[1]: Entering directory '/home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0d978.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD -o CMakeFiles/cmTC_0d978.dir/src.c.o -c /home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_0d978
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0d978.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD CMakeFiles/cmTC_0d978.dir/src.c.o -o cmTC_0d978 
/usr/bin/ld: CMakeFiles/cmTC_0d978.dir/src.c.o: in function `main':
src.c:(.text+0x2f): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x3b): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x47): undefined reference to `pthread_cancel'
/usr/bin/ld: src.c:(.text+0x58): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_0d978.dir/build.make:106: cmTC_0d978] Error 1
gmake[1]: Leaving directory '/home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:140: cmTC_0d978/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake cmTC_bc013/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_bc013.dir/build.make CMakeFiles/cmTC_bc013.dir/build
gmake[1]: Entering directory '/home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_bc013.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create -o CMakeFiles/cmTC_bc013.dir/CheckFunctionExists.c.o -c /usr/share/cmake-3.18/Modules/CheckFunctionExists.c
Linking C executable cmTC_bc013
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_bc013.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create CMakeFiles/cmTC_bc013.dir/CheckFunctionExists.c.o -o cmTC_bc013  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_bc013.dir/build.make:106: cmTC_bc013] Error 1
gmake[1]: Leaving directory '/home/*****/Documents/CAD/FreeCAD/FreeCAD/build_0.21/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:140: cmTC_bc013/fast] Error 2
```

_pthreads_ is installed of course, and _FreeCAD_ compiles without problems on this computer in other versions. I didn't find the corrsponding source files, and it looks like some sort of debugging or test setup. But LGPL requires to provide the **EXACT** source code of which the program was compiled.

**2.) LGPL requires that all tooling, including all scripts, are provided to achieve the EXACT same executable program as distributed.** 

```
"Source code" for a work means the preferred form of the work for
making modifications to it.  For a library, complete source code means
all the source code for all modules it contains, plus any associated
interface definition files, plus the scripts used to control compilation
and installation of the library.
```

But while there are compilation instructions for self-compilig FreeCAD – which do and did work on other versions – there are no descriptions and tooling and scripts to produce the DMG, AppImage and EXE as distributed by the project. This is also true for all other FreeCAD versions : there are no instructions on how to obtain the binary files. This is illegal.

**3.) there is a worrying comment in file** _~/tests/lib/CMakeLists.txt_ : 

```
# Note: CMake support is community-based. The maintainers do not use CMake
# internally.
```

This is illegal. Whatever the "_maintainers_ " use **MUST** be distributed and all scripts be included

**4.) The project is requested to respect the LGPL license and provide full source-code and tooling**

According to the GitHub rules about license and copyright violations, the project has 24 hours to comply. I invite you to not take this threat lightly

### Full version info

```shell
doesn't compile therefore can't give version info. The URL of the downloaded files is 

https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-05-07:
All build scripts are public, even the build logs. https://github.com/FreeCAD/freecad-feedstock https://github.com/FreeCAD/FreeCAD-Bundle/
calling this an lgpl violation is absurd

adrianinsaval, 2024-05-07:
> **3.) there is a worrying comment in file** _~/tests/lib/CMakeLists.txt_ :
> 
> ```
> # Note: CMake support is community-based. The maintainers do not use CMake
> # internally.
> ```

this is third party code and there's nothing to indicate that they aren't sharing tooling to build gtest, it seems to me that they use bazel. This is no cause for concern

kadet1090, 2024-05-07:
1.) I downloaded the source in tar.gz from the page you mentioned and apart from my distro forcing python3.12 that disables ability to build against Qt5 everything which requires separate patch everything works fine so check your setup.

yorikvanhavre, 2024-05-07:
I'll test the source provided with the release tomorrow. But that source package is provided by github itself, it IS the code that is used to build the release. 

Come on @Zolko-123 , what are those threats? You know we all zeal about the LGPL license.

You have a point regarding build scripts, though, they are indeed hard to find and not optimally documented, but things have improved a lot already, see https://github.com/FreeCAD/DevelopersHandbook

closed 2024-05-07 by yorikvanhavre

Zolko-123, 2024-05-28:
My mistake